rerun-io/rerun #6693: Impossible to click on a unit-length tensor in the tensor view

teh-cmc, 2024-07-01:
The title bar steals the mouse cursor before you can ever select the tensor and/or the view itself.

https://github.com/rerun-io/rerun/assets/2910679/f726b503-def0-4fd9-9cdd-6e39c1092e28

Wumpf, 2024-07-01:
Actually selecting tensors is not a thing at all today, it just doesn't do anything in any case.

ntjohnson1, 2026-11-05:
With the tensor hover in 0.42 and tensor pan/zoom just landed to main this looks to be resolved. Demonstrating [1], [1,1], and [1,512] tensors

https://github.com/rerun-io/rerun/commit/ee88e60c94fce20f57a07665d3980757b422cb3a
https://github.com/rerun-io/rerun/commit/f5e6e58fd71631dc049905af0f7154afc8dd56bc

https://github.com/user-attachments/assets/e371f0f9-41bc-4f03-8ccb-b7d703721db9

closed 2026-11-05 by ntjohnson1