rerun-io/rerun #6695: Tensor view's `FillKeepAspectRatio` mode is buggy

teh-cmc, 2024-07-01:
Data disappears and reappears as you hover it or resize the viewer.

https://github.com/rerun-io/rerun/assets/2910679/c708cada-4512-4a6b-bddc-5279780d0c10

Wumpf, 2024-07-01:
the description sounds like we're missing a redraw on changing the property, but the capture looks like much more is broken? I can see there's a 1D tensor here, but I don't get that's overdrawn there upon mouse interaction

Not unlikely that aspect ratio is broken in particular for 1D tensor

teh-cmc, 2024-07-03:
100% repro:
```
git checkout main
pixi run -e examples py-build
pixi run -e examples python tests/python/release_checklist/check_1d_tensor_data.py

Go to the tensor view, switch the dimension mapping so that it goes vertical, see that part of the tensor is somehow missing.
The problem gets worse as your window's height grow larger.
```
[image]

Wumpf, 2024-07-03:
Got a very subtle repro now. Looks like this is the on-hover axis labels not fully showing
[image]

abey79, 2025-04-23:
What you're seeing is basically the axis label slightly overlapping the tensor.

[image]

That label disappears on hover, thus the dynamic behaviour.

The root cause is that this label is cropped to the tensor area, which doesn't make any sense.

[image]

closed 2025-04-30 by Wumpf