rerun-io/rerun #6702: Pop-ups very slowly shrink to their final size -- too slowly?

teh-cmc, 2024-07-01:
https://github.com/rerun-io/rerun/assets/2910679/78e4cd90-3845-44f2-b38d-ce594787312f

emilk, 2024-07-02:
I fail to reproduce

teh-cmc, 2024-07-02:
I found a 100% repro:
```
git checkout main
rm -rf ~/.local/share/rerun/blueprints/*
pixi run -e examples py-build
pixi run -e examples python tests/python/release_checklist/check_static_components_ui.py
```

Now select any of the Position2D components in the streams view:
[image]

emilk, 2024-07-02:
Is is because `egui_extras::Table` not respecting the `resizing_pass` flag

emilk, 2024-07-02:
This turned into a rabbit hole of improving the layout and sizing of `egui_extras::Table`…

* First step a refactor: https://github.com/emilk/egui/pull/4755
* Next step: coming soon

closed 2024-07-03 by emilk