rerun-io/rerun #6722: Hover pop ups accumulate indefinitely as frame times go up

teh-cmc, 2024-07-02:
I have a problem with pop ups that just accumulate as i accidentally hover over stuff and never go away unless a click somewhere or remove focus from the Rerun window.
This barely happens at 60+ FPS, but it becomes more and more common when the viewer slows down (making OPF very painful to navigate on my dedicated GPU).

[image]
[image]

emilk, 2024-07-02:
I think the code for "the mouse is moving away from the tooltip, so close it" is somehow failing when framerate is low

emilk, 2024-07-02:
Fix:
* https://github.com/emilk/egui/pull/4757

closed 2024-07-03 by emilk