rerun-io/rerun #6748: WSL: Rerun native crashes on launch

jprochazk, 2024-07-03:
On WSL2, the native viewer crashes immediately on launch with "Could not find wayland compositor".

```
$ rerun
[2024-07-03T15:03:02Z INFO  re_sdk_comms::server] Hosting a SDK server over TCP at 0.0.0.0:9876. Connect with the Rerun logging SDK.
[2024-07-03T15:03:02Z INFO  re_viewer] Forcing graphics backend to Gl.
Error: winit EventLoopError: os error at /usr/local/cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.29.9/src/platform_impl/linux/wayland/event_loop/mod.rs:81: Could not find wayland compositor -> os error at /usr/local/cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.29.9/src/platform_impl/linux/wayland/event_loop/mod.rs:81: Could not find wayland compositor
```

Tested on both rerun `0.16.1` and latest `main`.

### System information

```
Windows Subsystem for Linux (WSL2)
Build: 19045
Branch: vb_release
Release: Ubuntu 22.04.4 LTS
Kernel: Linux 5.15.153.1-microsoft-standard-WSL2
```

Wumpf, 2024-07-03:
fresh wsl install running `pixi run rerun` off main I get a similar, but not identical error:
```
Error: winit EventLoopError: os error at /home/andreas/.cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.29.15/src/platform_impl/linux/wayland/event_loop/mod.rs:80: The wayland library could not be loaded
```

Running `WAYLAND_DISPLAY= pixi run rerun` gives me
```Error: winit EventLoopError: the requested operation is not supported by Winit```

After that I realized I didn't install any of the packages we ask linux users to install on https://rerun.io/docs/getting-started/troubleshooting (pretty bad user experience that this isn't printed :/)
Once done, I can no longer repro the issue at all and `pixi run rerun` just works Oo.

Can you check what happens when you go through the dependencies on our troubleshooting page if you haven't already?

automatedmis-take, 2024-08-12:
> On WSL2, the native viewer crashes immediately on launch with "Could not find wayland compositor".
> 
> ```
> $ rerun
> [2024-07-03T15:03:02Z INFO  re_sdk_comms::server] Hosting a SDK server over TCP at 0.0.0.0:9876. Connect with the Rerun logging SDK.
> [2024-07-03T15:03:02Z INFO  re_viewer] Forcing graphics backend to Gl.
> Error: winit EventLoopError: os error at /usr/local/cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.29.9/src/platform_impl/linux/wayland/event_loop/mod.rs:81: Could not find wayland compositor -> os error at /usr/local/cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.29.9/src/platform_impl/linux/wayland/event_loop/mod.rs:81: Could not find wayland compositor
> ```
> 
> The documented workaround of setting `WAYLAND_DISPLAY=` works, but we could at least try to improve the error message.
> 
> Tested on both rerun `0.16.1` and latest `main`.
> 
> ### System information
> ```
> Windows Subsystem for Linux (WSL2)
> Build: 19045
> Branch: vb_release
> Release: Ubuntu 22.04.4 LTS
> Kernel: Linux 5.15.153.1-microsoft-standard-WSL2
> ```

What are we supposed to set `WAYLAND_DISPLAY=` to? Where is this documented workaround? I'm running Ubuntu 24.04 LTS which is using the same kernel as your setup, and I got the same os error.

jprochazk, 2024-08-12:
It's documented in our troubleshooting guide: https://rerun.io/docs/getting-started/troubleshooting#wsl2