rerun-io/rerun #6757: Make (eye) camera speed configurable

Wumpf, 2024-07-04:
One of the many things that should go into the camera's blueprint!

See also
* https://github.com/rerun-io/rerun/issues/6013
* https://github.com/rerun-io/rerun/issues/5451

Parskatt, 2024-07-04:
Imo, it would be nice to just single press shift to increase and ctrl to decrease speed, and make them accumulate.

Parskatt, 2024-07-04:
But also setting via blueprint is nice, I have somewhat weird scenes where the autosetting is way too fast in general.

Wumpf, 2025-06-24:
Thanks to @jccampagne  speed is now configurable as a ephemeral view state in the view selection!
Keeping this open since we'd still want to persist things in the blueprint together with other camera properties

Wumpf, 2025-08-15:
* https://github.com/rerun-io/rerun/pull/10445

added it to the blueprint!