rerun-io/rerun #6761: Can't select/click a 2D primitive without also selecting underlying image

emilk, 2024-07-04:
When I hover a 2D primitive, like a box or a point, I expect only to hover that thing, not also all images under it.

[image]

Affects 2D spatial views, and maybe 3D ones too.

nikolausWest, 2024-07-04:
I don't actually agree with this expectation. If I run an object detector on an image, the edges of the bounding boxes are the most interesting parts of the image to inspect. If we only hover one thing at a time, it makes it super hard to look closely at around those borders. We used to have it that we only hovered one entity at a time but changed that because it made it really hard to debug for instance detections on images

emilk, 2024-07-04:
I can understand it for hovering, but for clicking it becomes quite confusing, and will become even more confusing when we have solid primitives. Say you click the middle of a solid-filled rectangle. Then you see an image in the selection view. WTH!?

In either case, I suggest we don't close this until we explain in the code (and maybe in the UI) that this behavior is intentional.

nikolausWest, 2024-07-04:
Fair enough. I think it would be pretty reasonable for clicking to just select the topmost thing but hovering going all the way through

abey79, 2024-12-09:
I suggest we bite the bullet and roll a heuristic-y approach here.

1) For hover, don't change the current behaviour (aka _everything_ that's under the mouse gets highlighted).

2) For click/cmd-click:
  - if the picked items are all image-like, select them all
  - if the picked items are a mix of image-like and primitive-like (point, line, etc.), ignore the image-like ones and select only the primitive-like ones

closed 2024-12-11 by emilk