rerun-io/rerun #6762: Numpy 2.0 Support

Serafadam, 2024-07-04:
**Is your feature request related to a problem? Please describe.**
Currently Rerun pip install requires Numpy 1.x, which prevents installing when Numpy 2.0 is already installed and some parts like logging images break after installing rerun-sdk with lower version and later upgrading it.

**Describe the solution you'd like**
Adding support for Numpy 2.0

emilk, 2024-07-04:
What is needed is a change to https://github.com/rerun-io/rerun/blob/9b0c4a22834fe6679dc92773efcdad59927815d9/rerun_py/pyproject.toml#L17

…and testing

Wumpf, 2024-07-19:
Via https://github.com/rerun-io/rerun/issues/6948:

> 
> Numpy 2.0 has been out for a bit now and is starting to show up as a real dep in more projects.
> 
> See: https://numpy.org/devdocs/numpy_2_0_migration_guide.html
> 
> Historically we had to limit ourselves to < 2.0 because of:
> - https://github.com/PyO3/rust-numpy/issues/409
> 
> However, I don't believe we depend on numpy bindings through PyO3 at this point so we may be able to relax our dep.

preston-thomson-dedrone, 2024-10-03:
@Wumpf Are you guys likely to update to the latest version of numpy in the next release?

Wumpf, 2024-10-03:
@preston-thomson-dedrone no we didn't get to this one so far

closed 2025-03-04 by jleibs