rerun-io/rerun #6784: Any plans for fully supporting ROS 2 visualization similar to Foxglove?

kminoda, 2024-07-05:
**Is your feature request related to a problem? Please describe.**

I am from Autoware community, which is an open sourced autonomous driving software based on ROS 2.
We use RViz and sometimes Foxglove on a daily basis, but the former one is not a modern tool and the latter one is announced to become closed.

**Describe the solution you'd like**

Thus, including me, developers from ROS 2 and especially Autoware community are very interested in rerun for similar visualization feature, such as runtime visualization and offline visualization from ROSBAG with seek bar UI.

**Describe alternatives you've considered**

As stated above, we are using RViz and Foxglove. But RViz is not the best solution for some of our usecases, and Foxglove is going to be closed-sourced.

**Additional context**

N/A

emilk, 2024-07-05:
Yes, ROS2 is on our roadmap:

https://github.com/rerun-io/rerun/blob/776e2e904c38e1ebdba018c7f7969e8d6a0238f7/docs/content/reference/roadmap.md#L34-L38

In the meantime I can refer you to https://rerun.io/docs/howto/ros2-nav-turtlebot

jbcpollak, 2024-09-05:
following.... Specifically hoping for rosbag and/or MCAP support. Anything to load archives from ROS pub/sub.

bradday4, 2024-09-10:
Hello Rerun, just wanted to start off by saying that the tool is amazing and superb for our visualization needs.

I'm wondering though what exactly does Official ROS2 bridge entail? Are there more specifics that could be provided? For example, replay from rosbag2, or mcap files, subscribing directly to ros topics, publishing its own topics, tf2 support, launch file support, etc...? 

appreciate the feedback :smile:

karimo87, 2024-09-12:
A quick update: we're working on a production grade ROS bridge in Rust over [here](https://github.com/rerun-io/rerun-ros). We're working hard to get it ready to handle more contributors within the next weeks, but we're not there yet.