rerun-io/rerun #6794: [Rerun Viewer] Save blueprint... doesn't open file dialog inside Docker container.

Tuntenfisch, 2024-07-05:
**Describe the bug**
I have a VS Code Dev Container set up with Rerun. Everything works fine except for the fact that any action inside the Rerun viewer that normally opens a file dialog does nothing.

**To Reproduce**
Steps to reproduce the behavior:
1. Open an empty folder in VS Code
2. Create `.devcontainer/devcontainer.json`:
```
{
	"build": {
		"dockerfile": "dockerfile",
		"context": ".."
	},
	"initializeCommand": [".devcontainer/initialize_command"],
	"runArgs": ["--network=host", "--runtime=nvidia"]
}
```
3. Create `.devcontainer/dockerfile`:
```
FROM ubuntu:24.04

SHELL ["/bin/bash", "-c"]

RUN apt-get update && apt-get -y install --no-install-recommends \
    sudo

RUN echo ubuntu ALL=\(root\) NOPASSWD:ALL > /etc/sudoers.d/ubuntu && \
    chmod 0440 /etc/sudoers.d/ubuntu

USER ubuntu

RUN sudo apt-get update && sudo apt-get install -y --no-install-recommends \
        libclang-dev \
        libatk-bridge2.0 \
        libfontconfig1-dev \
        libfreetype6-dev \
        libglib2.0-dev \
        libgtk-3-dev \
        libssl-dev \
        libxcb-render0-dev \
        libxcb-shape0-dev \
        libxcb-xfixes0-dev \
        libxkbcommon-dev \
        patchelf \
        vulkan-tools \
        mesa-vulkan-drivers \
        libxkbcommon-x11-0

RUN sudo apt install -y --no-install-recommends \
        python3-dev \
        python3-pip && \
    pip install --break-system-packages \
        rerun-sdk

ENV NVIDIA_DRIVER_CAPABILITIES=all
ENV NVIDIA_VISIBLE_DEVICES=all
```
4. Create `.devcontainer/initialize_command`:
```
#!/usr/bin/env bash

setup_xhost_forwarding() {
  # Ensure docker can forward GUI applications.
  if [[ -x "$(command -v xhost)" ]]; then
    xhost +local:docker > /dev/null
  fi
}

setup_xhost_forwarding
```

5. Open the VS Code command palette and start the container via `> Dev Containers: Reopen in Container`
6. Inside the container, open a terminal and execute `rerun`
7. Inside the opened Rerun viewer, select an action from the top left menu such as "Save blueprint..." or "Open...".
8. Observe that nothing happens.

**Expected behavior**
When triggering an action inside the Rerun viewer that normally opens up a file dialog, e.g. "Save blueprint..." or "Open...", a file dialog should open up. 

**Actual behavior**
No file dialog opens up. The action simply fails silently.

**Desktop:**
Output of `lsb_release -a`:
```
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.4 LTS
Release:	22.04
Codename:	jammy
```

**Rerun version**
```
rerun-cli 0.16.1 [rustc 1.76.0 (07dca489a 2024-02-04), LLVM 17.0.6] x86_64-unknown-linux-gnu 6447/merge d24e449, built 2024-05-29T15:06:39Z
```

**Additional context**
Output inside the terminal when starting Rerun:
```
[2024-07-05T21:47:46Z INFO  re_sdk_comms::server] Hosting a SDK server over TCP at 0.0.0.0:9876. Connect with the Rerun logging SDK.
[2024-07-05T21:47:47Z INFO  winit::platform_impl::platform::x11::window] Guessed window scale factor: 1
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
[2024-07-05T21:47:47Z INFO  egui_wgpu] There were 2 available wgpu adapters: {backend: Vulkan, device_type: DiscreteGpu, name: "NVIDIA GeForce RTX 4070 Laptop GPU", driver: "NVIDIA", driver_info: "535.183.01", vendor: 0x10DE, device: 0x2820}, {backend: Vulkan, device_type: Cpu, name: "llvmpipe (LLVM 17.0.6, 256 bits)", driver: "llvmpipe", driver_info: "Mesa 24.0.5-1ubuntu1 (LLVM 17.0.6)", vendor: 0x10005}
```

Tuntenfisch, 2024-07-06:
Okay, I'm not familiar at all with Rust but I compiled from source and found out that if set the rfd dependency inside [`rerun-io/rerun/Cargo.toml`](https://github.com/rerun-io/rerun/blob/main/Cargo.toml#L203C1-L203C80) to:
```
rfd = { version = "0.12", default-features = false, features = ["gtk3"] }
```
a file dialog opens. So it seems to work if the backend is `gtk3` and fails if the backend is `xdg-portal`. But I have no idea what to do with this information. Can I change something in my Docker container to make it work with `xdg-portal?

Tuntenfisch, 2024-07-06:
Okay, I've noticed changing the `rfd` dependency to
```
rfd = { version = "0.14.1", default-features = false, features = ["async-std", "xdg-portal"] }
```
allows the file dialog to open just fine inside the container while still using `xdg-portal` as a backend. Not sure of the implications of bumping the `rfd` version, tho...

emilk, 2024-07-07:
Bumping `rfd` sounds good to me! You wanna make a PR?

Tuntenfisch, 2024-07-07:
> Bumping `rfd` sounds good to me! You wanna make a PR?

Okay, will do. After some more testing I noticed that rfd version 0.14.1 works because of https://github.com/PolyMeilex/rfd/commit/ad3cdef63558924b75594676dfc1cc5c6a7b2890 which added a zenity fallback for xdg-portal. So I additionally need to have zenity installed inside my Docker container (which I don't mind).

emilk, 2024-11-02:
* Closed by https://github.com/rerun-io/rerun/pull/7953

closed 2024-11-02 by emilk