rerun-io/rerun #6801: Context menu width depends seemingly random on where it was clicked

Wumpf, 2024-07-08:
I'm not sure how I exactly got into this state, but the context appeared sometimes wide and sometimes narrow depending on the item:
https://github.com/rerun-io/rerun/assets/1220815/90aba2d6-8dcd-464f-af5a-73ce23bb1dba
Once (not captured on video) I got an even narrower context menu which had only space for 1-2 characters.

(0.17rc2 @ Windows)

Wumpf, 2024-07-08:
The same seems to happen with tooltips
[image]

emilk, 2024-07-08:
There's nothing random in your video - it seems to depend 100% on which element was clicked

Wumpf, 2024-07-08:
Not in the video, itself but
* it's unclear what's special about the second item
* The behavior hasn't been consistent over my overall Rerun session

emilk, 2024-07-08:
I could see this bug happening if the context menu changes content over time. If a given context menu is narrow the first time it is shown, it will remember this narrowness and then in the future break new elements to this width.

If this is indeed the problem, there are few different fixes to consider:
A) change the `Id` of the context menu based on its content
B) remove line wrapping from context menus

emilk, 2024-09-25:
I suspect this only happens when pixels-per-point is non-integer, due to some rounding problem, perhaps same problem as in https://github.com/emilk/egui/issues/5084.

The idea would be: we lay out the contents of the popup with a wide `max_size`/wrap-width. The text does not wrap, and the non-wrapped size of the text is reported up the `Ui` chain and the final size is then stored for the `Area`. This size is then set as the `max_size` on subsequent frames, but due to some rounding error, the text will now wrap, despite the width presumably being _exactly_ the width where it does not need to wrap.

I've tried to avoid this issue in epaint/egui by rounding text widths up to the nearest integer point width, but obviously there is still some bug somewhere. I'm investigating…

emilk, 2024-09-25:
I think this is the cause of the issue:
* https://github.com/emilk/egui/pull/5161#pullrequestreview-2327662175

emilk, 2024-09-25:
Repro inside of Rerun:

[image]

This happens both with `pixels_per_point=1` and `2`

emilk, 2024-09-25:
I think the original issue was actually solved by
* https://github.com/emilk/egui/pull/5161#pullrequestreview-2327662175

And that the tooltip issue is a separate one:
* https://github.com/emilk/egui/issues/5167