rerun-io/rerun #6811: MVP: python API to get dataframes out of an RRD

jleibs, 2024-07-08:
Needs to support our basic join semantics.

Should return data natively as pyarrow types -- does not need to convert to wrapped Rerun type objects.

Still need to think about the shape of the data result. There are a few different kind of results:
 - LatestAt: Dataframe where each row = index.
 - Range:
   - For non-batch (scalar) types this might be a dataframe where each row is a timepoint
   - For batch types this might be an iterator where each iteration is a LatestAt-style dataframe

nikolausWest, 2024-08-27:
@teh-cmc: does this issue have enough specificity to implement these queries or do you need further details on the exact semantics of the queries?

jleibs, 2024-08-27:
Depends on: https://github.com/rerun-io/rerun/issues/7284

closed 2024-10-04 by jleibs