rerun-io/rerun #6826: [tracking] More configurable view properties

Wumpf, 2024-07-09:
3D View
* https://github.com/rerun-io/rerun/issues/6825
    * https://github.com/rerun-io/rerun/issues/6013
    * https://github.com/rerun-io/rerun/issues/2638
    * https://github.com/rerun-io/rerun/issues/6757
* https://github.com/rerun-io/rerun/issues/2638
* https://github.com/rerun-io/rerun/issues/872
* https://github.com/rerun-io/rerun/issues/2663
    * (parts of it should be a view property)

2D View
* https://github.com/rerun-io/rerun/issues/6247

Plotting (BarChar + TimeSeriesView):
* https://github.com/rerun-io/rerun/issues/4818
* https://github.com/rerun-io/rerun/issues/6265
* https://github.com/rerun-io/rerun/issues/6773

Log:
* https://github.com/rerun-io/rerun/issues/6824

Time
* https://github.com/rerun-io/rerun/issues/5577
* https://github.com/rerun-io/rerun/issues/1381

Parskatt, 2024-09-17:
Want to bump specifically selecting EyeMode a'la https://github.com/rerun-io/rerun/blob/9fe16ba57c2f83f0644f9d7624384b047d7fe282/crates/viewer/re_space_view_spatial/src/ui.rs#L163-L167 blueprintly. I basically always want the first person viewer.

abey79, 2025-03-03:
I'll probably use the "blueprintify" keyword when searching for this issue, so with the present comment, I might actually find it.