rerun-io/rerun #6835: Add troubleshooting or how-to guide related to running Rerun in a Docker container

jleibs, 2024-07-09:
This keeps coming up and even if we don't officially support it, would be good to set expectations and point people in the right direction.

At time of writing, with 0.17 the following works.

Install `nvidia-container-runtime`.

Make sure `/etc/docker/daemon.json` contains:
```
{
  "runtimes": {
    "nvidia": {
      "args": [],
      "path": "nvidia-container-runtime"
    }
  }
}
```

Make sure your docker image includes at least: `libgtk-3-dev libxkbcommon-x11-0 vulkan-tools`

Example `Dockerfile` for an ubuntu image.
```
FROM ubuntu:22.04

RUN apt-get update && apt-get install -y python3-pip libgtk-3-dev libxkbcommon-x11-0 vulkan-tools
RUN python3 -m pip install rerun-sdk==0.17.0
```

Jump through all the hoops necessary for GPU access, X authentication, etc.

In particular the following two seem to be crucial:
- `--runtime=nvidia`
- `-e NVIDIA_DRIVER_CAPABILITIES=all`

```
XSOCK=/tmp/.X11-unix
XAUTH=/tmp/.docker.xauth
xauth nlist $DISPLAY | sed -e 's/^..../ffff/' | xauth -f $XAUTH nmerge -
chmod 777 $XAUTH
docker run --runtime=nvidia --rm --gpus all -it --privileged --network=host -e NVIDIA_DRIVER_CAPABILITIES=all -e DISPLAY=$DISPLAY -v $XSOCK:$XSOCK -v $XAUTH:$XAUTH -e XAUTHORITY=$XAUTH rerun:0.17.0 rerun
```

Hillowwold, 2024-07-15:
Hi! I am able to display the Rerun viewer when I run this code on my ssh server, but the viewer is extremely slow and reaches the point where it is not possible. Is there any way to improve this please?

jleibs, 2024-07-15:
> Hi! I am able to display the Rerun viewer when I run this code on my ssh server, but the viewer is extremely slow and reaches the point where it is not possible. Is there any way to improve this please?

Can you clarify, are you trying to run the viewer on a remote machine, and then tunnel the viewer over SSH, such as via X-forwarding?  I would not expect this to work. The viewer is fairly graphics intensive and needs access to the local GPU.

To run on a remote machine, I would still recommend running the viewer locally, and then connecting to it from the client using the `.connect()` API. See: https://rerun.io/docs/reference/sdk-operating-modes

Divelix, 2025-02-24:
I just found out that if I put
```docker
RUN apt update && apt install -q -y --no-install-recommends \
    libgtk-3-dev \
    libxkbcommon-x11-0 \
    vulkan-tools
```
inside Dockerfile, I get error on `rerun` command:
```log
[2025-02-24T09:37:29Z INFO  winit::platform_impl::linux::x11::window] Guessed window scale factor: 1
[2025-02-24T09:37:29Z WARN  wgpu_hal::vulkan::instance] Unable to find extension: VK_KHR_surface
[2025-02-24T09:37:29Z WARN  wgpu_hal::vulkan::instance] Unable to find extension: VK_KHR_xlib_surface
[2025-02-24T09:37:29Z WARN  wgpu_hal::vulkan::instance] Unable to find extension: VK_KHR_xcb_surface
[2025-02-24T09:37:29Z WARN  wgpu_hal::vulkan::instance] Unable to find extension: VK_KHR_wayland_surface
[2025-02-24T09:37:29Z WARN  wgpu_hal::vulkan::instance] Unable to find extension: VK_EXT_swapchain_colorspace
[2025-02-24T09:37:29Z WARN  wgpu_hal::vulkan::instance] Unable to find extension: VK_KHR_get_physical_device_properties2
libEGL warning: DRI3: Screen seems not DRI3 capable
libEGL warning: DRI2: failed to authenticate
[2025-02-24T09:37:29Z ERROR eframe::native::run] Exiting because of error: app creation error: The GPU/graphics driver is lacking some abilities: Adapter does not support drawing to texture format R32Float. Check the troubleshooting guide at https://rerun.io/docs/getting-started/troubleshooting and consider updating your graphics driver.
Error: app creation error: The GPU/graphics driver is lacking some abilities: Adapter does not support drawing to texture format R32Float. Check the troubleshooting guide at https://rerun.io/docs/getting-started/troubleshooting and consider updating your graphics driver.
```
But if I install these 3 packages manually in creted container shell, rerun works fine. Why could that be?

Divelix, 2025-02-24:
I used latest 0.22.1 version, after downgrade to 0.17.0 it runs fine without manual apt install.

Wumpf, 2025-03-03:
@Divelix try setting the software rasterizer (or better an actual driver) explicitly via an ICD file as described [here](https://rerun.io/docs/getting-started/troubleshooting#multiple-gpus)
0.17.0 will likely run into some runtime error once you try to interact with anything in 2D or 3D views.

jleibs, 2025-03-04:
@Divelix I have just confirmed that the instructions listed in this issue continue to work for me on 0.22.1 (and have updated accordingly).

Are you using that exact Dockerfile and run command?

In general, the best way to get support on this is to include as much information as possible:
- The host OS you are using
- The docker version you are using
- The nvidia-container-runtime version you are using
- A copy of the contents of the Dockerfile
- The exact `docker run` invocation you are using

Divelix, 2025-03-05:
@jleibs interesting, I had to use `prime-select nvidia` previously, now switched to `prime-select on-demand` and your Dockerfile+run command worked fine for both 0.17.0 and 0.22.0 - gui starts without errors. It seems like dockerized rerun requires intergated gpu for some reason, while rerun on host doesn't.

My system:
- `linux ... 6.8.0-52-generic #53~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Jan 15 19:18:46 UTC 2 x86_64 x86_64 x86_64 GNU/Linux`
- `Docker version 28.0.1, build 068a01e`
- `Driver Version: 535.183.01   CUDA Version: 12.2`
- nvidia-container-runtime:
```NVIDIA Container Runtime version 1.17.4
commit: 9b69590c7428470a72f2ae05f826412976af1395
spec: 1.2.0

runc version 1.2.4
commit: v1.2.4-0-g6c52b3f
spec: 1.2.0
go: go1.22.10
libseccomp: 2.5.3
```
- Dockerfile is the same as in your example
- Run invocation the same as yours

jlack1987, 2025-03-29:
I'm also unable to run it in a docker container it seems. Not sure if the following console messages are useful, but when I run my (c++) application it gives,
```
Error: winit EventLoopError: os error at /usr/local/cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.30.7/src/platform_impl/linux/mod.rs:787: Failed to load one of xlib's shared libraries
[2025-03-29T17:33:07Z WARN  re_sdk_comms::buffered_client] Failed to send message after 3 attempts: Failed to connect to Rerun server at 127.0.0.1:9876: Connection refused (os error 111)
[2025-03-29T17:33:10Z WARN  re_sdk_comms::buffered_client] Dropping messages because tcp client has timed out.
[2025-03-29T17:33:10Z WARN  re_sdk_comms::buffered_client] Dropping messages because tcp client has timed out.
[2025-03-29T17:33:10Z WARN  re_sdk_comms::tcp_client] Tried to flush while TCP stream was still Pending. Data was possibly dropped.
```