rerun-io/rerun #6849: Zoom and detail popup for tensor view

roym899, 2024-07-10:
**Is your feature request related to a problem? Please describe.**
I'd like to inspect some values in a tensor, however, the tensor view doesn't allow zooming and also has no detail popup on hover like the image view does. 

There is a workaround to inspect the tensor data with a detail popup by selecting the tensor in the blueprint panel and inspect it in the tiny preview in the Selection panel, but it's too tiny to be useful for larger tensors and only works for image-like tensors.

**Describe the solution you'd like**
Support zoom in the tensor view and add a detail popup similar to the image view to be able to inspect the actual value in the tensor.

**Related issues**
- #1630 
- #1510

OskarElek, 2024-10-24:
Seconding this - it would be an extremely useful feature @Wumpf 
Pretty much same dialog as for images/depth/videos that would work in tensor view and allow to inspect the actual numerical values.

ntjohnson1, 2026-11-05:
Hover landed in the last release. Zoom just landed to main so it should be in the next https://github.com/rerun-io/rerun/commit/f5e6e58fd71631dc049905af0f7154afc8dd56bc

closed 2026-11-05 by ntjohnson1