rerun-io/rerun #6865: cargo install rerun-cli fails for v0.17.0

andresv, 2024-07-11:
On macOS with rust 1.79.0:

```
rustc --version
rustc 1.79.0 (129f3b996 2024-06-10)

cargo install rerun-cli
    Updating crates.io index
  Installing rerun-cli v0.17.0
    Updating crates.io index
     Locking 618 packages to latest compatible versions
      Adding accesskit v0.12.3 (latest: v0.16.0)
      Adding accesskit_consumer v0.16.1 (latest: v0.24.0)
      Adding accesskit_macos v0.10.1 (latest: v0.17.0)
....

   Compiling re_types_core v0.17.0
error: Parse error while parsing value description: unbalanced source
  --> /Users/andres/.cargo/registry/src/index.crates.io-6f17d22bba15001f/re_types_core-0.17.0/src/lib.rs:15:10
   |
15 | #![doc = document_features::document_features!()]
   |          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = note: this error originates in the macro `document_features::document_features` (in Nightly builds, run with -Z macro-backtrace for more info)

   Compiling arboard v3.4.0
   Compiling fsevent-sys v4.1.0
   Compiling filetime v0.2.23
error: could not compile `re_types_core` (lib) due to 1 previous error
warning: build failed, waiting for other jobs to finish...
error: failed to compile `rerun-cli v0.17.0`, intermediate artifacts can be found at `/var/folders/5b/jlxfj70x1fb_n2l6qh21xxfh0000gn/T/cargo-installTQB5ME`.
To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.
```

emilk, 2024-07-11:
Can confirm 😬

emilk, 2024-07-11:
The fix is to add `--locked`:

```
cargo install rerun-cli --locked
```

emilk, 2024-07-11:
Upstream issue: https://github.com/slint-ui/document-features/issues/26

emilk, 2024-07-11:
Add recommendation to use `--locked` here: https://github.com/rerun-io/rerun/pull/6868

emilk, 2024-07-11:
The faulty `document-features` version has been yanked: https://crates.io/crates/document-features/0.2.9

Thanks for reporting this to us!

closed 2024-07-11 by emilk