rerun-io/rerun #6889: Tagged components

emilk, 2024-07-15:
Consider the `Color` component as applied to a `Mesh3D` archetype. Does the `Color` specify the color of faces, edges, or vertices?

To support this, we could introduced _tagged_ components, allowing each entity to have multiple instances of the same component, if they have different tags.

Each `(Tag, ComponentName)` tuple would be its own column of data.

## References
https://www.flecs.dev/flecs/md_docs_2Relationships.html

Wumpf, 2024-07-22:
An instance of tagged component @jleibs and me discussed recently is fill out and wireframe color in a situation where both solid & fill are supported at the same time (another outcome of the discussion was that we don't want to support it right now, more on that later & elsewhere):
Let's say we have a visualizer that supports doing both at the same time:
* if only one of wireframe & fill is enabled: query _any_ color and use that
* if both are enabled: check for specific color tags and act accordingly
alternatively the tags themselves can also influence the default of what's enabled!

kpreid, 2024-07-28:
This seems like a potentially lot of added fundamental complexity in implementation and possibly API. I wonder if there's a way to get the same effect, simpler, in addition to wondering exactly what the semantics of this tag system is going to be.

* What if component types as currently known were removed entirely, and Rerun used only pairs of `(Tag, Data)`, that is, components would be like `(SolidColor, Rgba)` and `(Radius, f32)`? That's still a unique static type per component.

* Another, probably more Rust-only frame but this is a brainstorming sort of remark: what if these were just parameterized components? That is, `components::Color<Solid>` vs `components::Color<Edge>`. That way, you avoid code duplication, and don't need to extend the existing architecture of component types (but probably need fancier code generation).

teh-cmc, 2024-07-29:
We've had a long discussion about leaf (nee "out-of-tree") transforms a couple days back, during which the desire for tagged components came up _a lot_. Again. We keep running into data modeling deadlocks.

In parallel, @gavrelina  has been experimenting with the Dataframe View and has hit more of the same issues.

We keep running into data modeling deadlocks, again and again.

---

My guess is that tagged components will greatly influence the design of our data model and query language, and to an extent even the UX of the viewer itself. As such I think it's really important that we get there sooner rather than later.
It doesn't have to be perfect either: the sooner we get to play with these new tools and refine our data model with them in mind, the sooner we can start stabilizing and ultimately _(gasp)_ standardizing things.

It would also be quite nice to be completely done with major ABI breaking changes ASAP, before we enter the :sparkles: disk-based era :sparkles:.

Thus, here's a quick proposal to move us towards that goal.

---

## Context

Today, the atomic unit of data in Rerun is a Chunk column.

A Chunk column is fully qualified by two things: a Rerun `ComponentName` and an Arrow `Datatype` (NOTE: I'm intentionally omitting the 3rd qualifier, the "Rerun datatype name", which is a vague internal concept that is not actually materialized most of the time. It doesn't matter for this discussion.):
- The component name (`Position3D`, `Color`, `Radius`, …) specifies the semantic of the data, "how things behave".
- The datatype (`[3]Float32`, `Uint32`, …) specifies the memory layout of the data.

The two are for the most part completely orthogonal to one another.

This information is all stored within the column metadata, and denormalized into the store indices as needed.

At runtime, a Rerun system will look for a piece of data by searching for the semantic its interested in, and then interpreting the returned data based on its datatype:
```rust
let data = store.latest_at("my_entity", TimeInt::MAX, "rerun.components.Position3D")?; // untyped (!)
let data = data.try_downcast::<&[[f32; 3]]>()?; // typed
```

All of this works pretty nicely, except for one major limitation: you cannot re-use the same semantics twice (or more) on a single entity.

Example: imagine an entity `"detections"` that holds both 3D points and 3D boxes at the same time: they will have to share a batch of `"rerun.components.Position3D"` (for `Points3D::positions` and `Boxes3D::centers`).
Sometimes this is exactly what you want, and things are great. Just enable both the `Points3D` and `Boxes3D` visualizers and you're good to go.
Sometimes it is not what you want, and you're doomed.

This problem infects everything and leads to all kinds of nasty data modeling deadlocks all over the place.

## Proposal

The core idea of this proposal is trivial: to replace the very limited Rerun `ComponentName` with a much more flexible `ComponentDescriptor`.
Of course, this has a ton of ripple effects. I'll try to cover the most important ones, but there's too many to cover them all.

We should be able to get there in small increments that can be merged as they come, with complete feature parity and no visible impact on end-users whatsoever.

Once we're there, we'll be able to start experimenting with all kinds of crazy ideas.

### Data model changes

A Chunk column would still be fully qualified by two bits of information: a Rerun `ComponentDescriptor` (semantics) and an Arrow `Datatype` (memory layout).
All that metadata is stored in the same place as before (column metadata and/or arrow schema).

A `ComponentDescriptor` would look like the following:
```rust
/// A [`ComponentDescriptor`] fully describes the semantics of a column of data.
pub struct ComponentDescriptor {
    /// Optional name of the `Archetype` associated with this data.
    ///
    /// `None` if the data wasn't logged through an archetype.
    ///
    /// Example: `rerun.archetypes.Points3D`.
    archetype_name: Option<ArchetypeName>,

    /// Semantic name associated with this data.
    ///
[... 306 lines cut ...]

## FAQ

### What becomes of indicator components?

They're gone; they're just redundant at that point: use the `archetype_name` field in the `ComponentDescriptor` instead.

### What about datatype conversions?

As far as I can tell, datatype conversions is a completely orthogonal problem.

"Tagged components" is about sharing semantics across columns, "datatype conversions" is about making it easy to change the memory layout of a column.

### What about blueprint defaults/overrides?

Mostly nothing changes, except now the blueprint has an opportunity to define a default value for all tags or a specific one, or both.

```rust
blueprint.set_default("*::rerun.components.Color#*", Color::Blue);
blueprint.set_default("*::rerun.components.Color#vertex_colors", Color::Green);
```

### What about the DataframeView?

The dataframe view now has all the information it needs to properly distinguish between data with similar semantics.

jleibs, 2024-07-29:
> As far as I can tell, datatype conversions is a completely orthogonal problem.

I disagree with with this. At most it's partially orthogonal. My big take away from the last year of working with our components and datatypes is that the line between semantics and datatypes is far from clear. I think this gives us an opportunity to improve that dramatically and we might as well plan to take advantage of it.

The previous strict hierarchy of component > datatype forced us into arbitrary distinctions. For example, consider color.

It was fairly clear that Color should be a component. But if color was the component, then the encoding of the color must (by necessity) be a datatype. This lead to datatypes like `Rgba32` and the implication that ultimately every color representation should be it's own datatype. But, at the end of the day, rgba32 is still just a sematic tag on top of a datatype primitive such as Vec4u8. Similar story with rotations. Is quaternion its own datatype, or is it a semantic label on top of vec4f?

Tags appears to give us flexible yet structured way to introduce more dimensionality to the way we talk about data. In particular, things like Color-spaces, Units, Encodings, etc. all probably fit much better into semantic tags than they do in the datatype system.

For me this path leads to, for example talking about rotations as:
- tags: (component: rotation, representation: quat_wxyz), datatype: vec4f
- tags: (component: rotation, representation: mat3x3), datatype: mat3x3d

We could theoretically try to have BOTH component-tags and datatype-tags, but I suspect the nitpicking over which kind of tag goes where would be endless while providing little-to-no practical value.

The logical end-point is that datatypes are JUST arrow schema aliases they ONLY talk about the shape of the data (e.g. Vec4f) and never talk about what that data represents (e.g. Quat4f). Everything else becomes a component-descriptor tag.

This generally means that "datatype conversions" are totally generic and follow very standard software equivalents:
- Cast a u8 to an f32.
- Extract the 3rd field of a Vec4x

A big chunk of the complexity previously attempted to datatype conversions now becomes "Component descriptor conversions." Same problem, new name.

teh-cmc, 2024-07-29:
Agree 100%

grtlr, 2024-10-22:
I've had a discussion with @abey79 today that touched upon this PR. For the graph primitives PR #7500, it would be great if we could log multiple components of the same data type to the same entity, possibly by discerning them with tags so that edges and nodes can live in the same entity while still allowing fields like `edge_colors` and `node_colors`.

I was wondering: How will that work with selections? If I understand correctly, we currently refer to "objects" in an entity by their [`InstancePath`](https://docs.rs/rerun/latest/rerun/external/re_entity_db/struct.InstancePath.html). This works because the main (reuqired) component dictates the index. Now if we have an arbitrary amount of components per entity wouldn't we need a special way to discern them too?

abey79, 2024-10-22:
We don't currently have a solution for this for archetypes which dont have an "instance-to-instance" match semantic. That's the case for the `Mesh3D` archetype: the instance index of the `Position3D` and `TriangleIndices` have no relationship (contrary to e.g. `Position3D` <-> `Color`). So if you select an instance path like this, you get unrelated values for `TriangleIndices` and `Position3D`:

[image]

I don't think we have an established plan for this going forward. We could have an archetype-level field flag (that gets propagated as component tag) to indicate that "this component doesn't have instance-match semantics". It would be applied e.g. to `Mesh3D`'s `TriangleIndices` or, for the Graph view, the `Edge` component. In the latter case, we'd want to go even further and tag a subset of components as having a shared instance index semantic (`Edge`, `EdgeColor`, `EdgeLabel`, etc.). Maybe? But that's future truth and will have to start without for now.