rerun-io/rerun #6901: Fix `segment_anything_model` example

emilk, 2024-07-16:
In https://github.com/rerun-io/rerun/pull/6900 I disabled the `segment_anything_model` example from building on CI, because it is broken.

The `segment_anything_model` example exits with a non-zero, without explaining why. It's this that does it:
https://github.com/rerun-io/rerun/blob/27b48c3333ef15bfab763b59dcd106bc6b49d826/examples/python/segment_anything_model/segment_anything_model.py#L78

Repro:

```
pixi run -e examples py-build-examples
pixi run -e examples python3 -m segment_anything_model --save example_data/segment_anything_model.rrd
```

The CI is running
```
pixi run -e examples build-examples rrd --channel nightly example_data 
```

Note that `py-build-examples` is not the same as `build-examples`. It's confusing.

jleibs, 2024-07-18:
In the future, it would be very helpful to:
1. Cite the specific action.yml file that is failing.
2. Link to a failed build run for additional context.

jleibs, 2024-08-02:
These all run for me now. Closing this issue. Folks can open new issues with concrete repros as necessary.

closed 2024-08-02 by jleibs