rerun-io/rerun #6931: Cannot install under WSL2 Ubuntu 20.04

bobpbbd104, 2024-07-17:
Tried the pip3 install, I get the error:

ERROR: Could not find a version that satisfies the requirement rerun-sdk (from versions: none)
ERROR: No matching distribution found for rerun-sdk

Not that it would matter considering the error, I also installed your suggested Linux dependencies. 

Then tried under Windows, but only get blank app window after running rerun.

Thanks
Robert

Wumpf, 2024-07-18:
Repros for me cleanly, Ubuntu 22 works fine for me. Which almost certainly means that we regressed on our glibc requirement again (see https://github.com/rerun-io/rerun/issues/1320)

The conda build is apparently not affected by this `conda install -c conda-forge rerun-sdk` worked for me and was able to run, you can use this as a workaround meanwhile.

To understand "blank app window"-issue better, can you please provide the following:
* how did you run it
* the log from commandline
* what gpu are you running, driver versions may be helpful as well
* when you say blank, do you mean an empty window? Probably easiest to post a screenshot :)

Thank you!

Wumpf, 2024-07-18:
Regarding the pip issue: Turns out this is _not_ a glibc issue, but something about the out of the box configuration of pip:
* `uv pip install rerun-sdk` works (https://astral.sh/blog/uv)
* `pipx install rerun-sdk` works as well
* after installing rerun with conda and then uninstalling, `pip` will also just work
No clue at this point what is going on, verbose logs weren't helpful either.

Wumpf, 2024-07-18:
turns out this is an issue with old pip versions, in this case `20.0.2` 
Running `pip install --upgrade pip` installs a new pip (`24.1.2` as of writing), careful however that this may not update the pip at `pip` command, in my case I had to run `pip3` then to get the right version afterwards

Wumpf, 2024-07-22:
@bobpbbd104 any more information you can provide on the windows blank screen?

Asherchi, 2024-08-27:
got the same problem in wsl Ubuntu18.04.

Wumpf, 2024-10-22:
I don't think there's much more we can do other than putting this into the troubleshooting guide since this seems to be an issue with old pip versions.

closed 2024-10-22 by Wumpf