rerun-io/rerun #6933: Fire-and-forget logging mode

teh-cmc, 2024-07-18:
It is pretty common to want to log data in a fire-and-forget kind of way, e.g. drop data if there is no viewer listening in when using `connect()`.

Surprisingly (to me, at least?), we do not seem to offer any way of doing this at the moment. Data always gets buffered either a the `RecordingStream` layer or in one the `Sink`s, there's just no way to drop it.

rgolovanov, 2024-08-07:
@teh-cmc could you clarify a bit the problem? when does it occur?

According to API documentation:
```
        /// flush_timeout_sec:
        /// The minimum time the SDK will wait during a flush before potentially
        /// dropping data if progress is not being made. Passing a negative value indicates no
        /// timeout, and can cause a call to `flush` to block indefinitely.
        ///
        /// This function returns immediately.
        Error connect(std::string_view tcp_addr = "127.0.0.1:9876", float flush_timeout_sec = 2.0)
            const;
```

the `flush_timeout_sec` should handle problems with connect and drop all accumulated buffer if the viewer wasn't connected within specified time limit.

teh-cmc, 2024-08-08:
`flush_timeout_sec = 0` should indeed work, but it simply didn't the last time I tried it.

Either there's a bug in its implementation, or there is something else entirely that is also buffering data.

DevonMorris, 2025-01-07:
Any updates on when you think this will land? We still periodically struggle with this

rrad5409, 2026-02-07:
Would like to mention this functionality is completely missing from the python sdk, as there are no timeout parameters to any of the sinks or connection functions.