rerun-io/rerun #6949: Error in Surface::configure: Validation Error: Not enough memory left

k-bx, 2024-07-19:
**Describe the bug**

`pixi run rerun` immediately crashes with "Not enough memory left" error. I believe I have quite a lot of memory, both RAM and GPU RAM free.

```
$ pixi run rerun
✨ Pixi task (rerun in default): cargo run --package rerun-cli --no-default-features --features native_viewer --
    Finished dev [optimized + debuginfo] target(s) in 0.16s
     Running `target/debug/rerun`
[2024-07-19T14:25:36Z DEBUG rerun::run] Detected 32 cores. Using 30 compute threads.
[2024-07-19T14:25:36Z DEBUG re_memory::memory_limit] Setting memory limit to 70.6 GiB, which is 75% of total available memory (94.1 GiB).
[2024-07-19T14:25:36Z INFO  re_sdk_comms::server] Hosting a SDK server over TCP at 0.0.0.0:9876. Connect with the Rerun logging SDK.
[2024-07-19T14:25:36Z DEBUG eframe] Using the wgpu renderer
[2024-07-19T14:25:36Z DEBUG eframe::native::wgpu_integration] Event::Resumed
[2024-07-19T14:25:36Z DEBUG eframe::native::file_storage] Loading app state from "/home/ziggy/.local/share/rerun/app.ron"…
[2024-07-19T14:25:36Z INFO  winit::platform_impl::platform::x11::window] Guessed window scale factor: 1
[2024-07-19T14:25:36Z DEBUG winit::platform_impl::platform::x11::window] Calculated physical dimensions: 1470x920
[2024-07-19T14:25:36Z DEBUG winit::platform_impl::platform::x11::ime] failed to create IME context without preedit support.
[2024-07-19T14:25:36Z WARN  wgpu_hal::vulkan::instance] InstanceFlags::VALIDATION requested, but unable to find layer: VK_LAYER_KHRONOS_validation
[2024-07-19T14:25:36Z WARN  wgpu_hal::vulkan::instance] GENERAL [Loader Message (0x0)]
    	terminator_CreateInstance: Failed to CreateInstance in ICD 3.  Skipping ICD.
[2024-07-19T14:25:36Z WARN  wgpu_hal::vulkan::instance] 	objects: (type: INSTANCE, hndl: 0x63975a7e2040, name: ?)
libEGL warning: DRI2: failed to authenticate
[2024-07-19T14:25:36Z INFO  egui_wgpu] There were 3 available wgpu adapters: {backend: Vulkan, device_type: DiscreteGpu, name: "NVIDIA GeForce RTX 4090", driver: "NVIDIA", driver_info: "550.54.15", vendor: 0x10DE, device: 0x2684}, {backend: Vulkan, device_type: Cpu, name: "llvmpipe (LLVM 15.0.7, 256 bits)", driver: "llvmpipe", driver_info: "Mesa 23.2.1-1ubuntu3.1~22.04.2 (LLVM 15.0.7)", vendor: 0x10005}, {backend: Gl, device_type: Cpu, name: "llvmpipe (LLVM 15.0.7, 256 bits)", driver: "OpenGL", driver_info: "4.5 (Compatibility Profile) Mesa 23.2.1-1ubuntu3.1~22.04.2", vendor: 0x10005}
[2024-07-19T14:25:36Z DEBUG egui_wgpu] Picked wgpu adapter: backend: Vulkan, device_type: DiscreteGpu, name: "NVIDIA GeForce RTX 4090", driver: "NVIDIA", driver_info: "550.54.15", vendor: 0x10DE, device: 0x2684

thread 'main' panicked at 'Error in Surface::configure: Validation Error

Caused by:
    Not enough memory left.
'
wgpu-0.20.1/src/backend/wgpu_core.rs:751
stack backtrace:
   6: core::panicking::panic_fmt
             at core/src/panicking.rs:72:14
   7: wgpu::backend::wgpu_core::ContextWgpuCore::handle_error_fatal
             at wgpu-0.20.1/src/backend/wgpu_core.rs:314:9
   8: <wgpu::backend::wgpu_core::ContextWgpuCore as wgpu::context::Context>::surface_configure
             at wgpu-0.20.1/src/backend/wgpu_core.rs:751:13
   9: <T as wgpu::context::DynContext>::surface_configure
             at wgpu-0.20.1/src/context.rs:2128:9
  10: wgpu::Surface::configure
             at wgpu-0.20.1/src/lib.rs:5096:9
  11: egui_wgpu::winit::Painter::configure_surface
             at egui-wgpu/src/winit.rs:177:9
  12: egui_wgpu::winit::Painter::resize_and_generate_depth_texture_view_and_msaa_view
             at egui-wgpu/src/winit.rs:338:9
  13: egui_wgpu::winit::Painter::add_surface::{{closure}}
             at egui-wgpu/src/winit.rs:306:9
      egui_wgpu::winit::Painter::set_window::{{closure}}
             at egui-wgpu/src/winit.rs:214:62
      pollster::block_on
             at pollster-0.3.0/src/lib.rs:128:15
  14: eframe::native::wgpu_integration::WgpuWinitApp::init_run_state
             at eframe/src/native/wgpu_integration.rs:203:13
      <eframe::native::wgpu_integration::WgpuWinitApp as eframe::native::winit_integration::WinitApp>::on_event
             at eframe/src/native/wgpu_integration.rs:427:21
  15: eframe::native::run::run_and_return::{{closure}}
             at eframe/src/native/run.rs:125:28
      core::ops::function::impls::<impl core::ops::function::FnMut<A> for &mut F>::call_mut
             at core/src/ops/function.rs:294:13
  16: winit::platform_impl::platform::x11::EventLoop<T>::single_iteration
             at winit-0.29.15/src/platform_impl/linux/x11/mod.rs:553:13
      winit::platform_impl::platform::x11::EventLoop<T>::pump_events
             at winit-0.29.15/src/platform_impl/linux/x11/mod.rs:441:13
      winit::platform_impl::platform::x11::EventLoop<T>::run_on_demand
             at winit-0.29.15/src/platform_impl/linux/x11/mod.rs:408:19
  17: winit::platform_impl::platform::EventLoop<T>::run_on_demand
             at winit-0.29.15/src/platform_impl/linux/mod.rs:829:56
      <winit::event_loop::EventLoop<T> as winit::platform::run_on_demand::EventLoopExtRunOnDemand>::run_on_demand
             at winit-0.29.15/src/platform/run_on_demand.rs:80:9
      eframe::native::run::run_and_return
             at eframe/src/native/run.rs:73:5
      eframe::native::run::run_wgpu::{{closure}}
             at eframe/src/native/run.rs:424:13
      eframe::native::run::with_event_loop::{{closure}}
             at eframe/src/native/run.rs:58:12
      std::thread::local::LocalKey<T>::try_with
             at std/src/thread/local.rs:270:16
      std::thread::local::LocalKey<T>::with
             at std/src/thread/local.rs:246:9
      eframe::native::run::with_event_loop
             at eframe/src/native/run.rs:48:16
      eframe::native::run::run_wgpu
             at eframe/src/native/run.rs:422:16
      eframe::run_native
             at eframe/src/lib.rs:268:13

Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting 
Report bugs: https://github.com/rerun-io/rerun/issues

```

**Rerun version**

Rerun from https://github.com/rerun-io/rerun/pull/6561 source

Wumpf, 2024-07-19:
Strange, haven't seen this one yet! Surface creation failing with out of memory I'd expect that it picks non-sensical render target size, but it says earlier `1470x920` with scale factor of 1.

Does it happen as well when you build [Rerun 0.17](https://github.com/rerun-io/rerun/tree/0.17.0)?

If you have the time to do so it would be interesting to check if the basic wgpu examples also exhibit this problem for you. For that you'd need to check out [wgpu](https://github.com/gfx-rs/wgpu) and run `cargo run --bin wgpu-examples hello_triangle`

k-bx, 2024-07-19:
Yup, same error on 0.17.

Just realised another thing which might be relevant: running things via xrdp + Microsoft Remote Desktop from my macOS to Ubuntu 22.04 machine.

k-bx, 2024-07-19:
Roger, thanks for digging into this!

WGPU does error:

```
ziggy@echo-gpu:~/workspace/wgpu$ cargo run --bin wgpu-examples hello_triangle
    Finished dev [unoptimized + debuginfo] target(s) in 0.08s
     Running `target/debug/wgpu-examples hello_triangle`
libEGL warning: DRI2: failed to authenticate
thread 'main' panicked at wgpu/src/backend/wgpu_core.rs:786:18:
Error in Surface::configure: Validation Error

Caused by:
  Not enough memory left.

note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

k-bx, 2024-07-19:
Made an issue https://github.com/gfx-rs/wgpu/issues/5993

Wumpf, 2024-07-19:
awesome, was just about to ask that. Thanks for helping to dig on this!