rerun-io/rerun #6977: Rerun crashes when saving

finnBsch, 2024-07-24:
**Describe the bug**
It seems that when the logged data becomes too big (like 12 Gb), rerun freezes when trying to save the recording and then crashes. No error outputs in the terminal, even with RUST_BACKTRACE and RUST_DEBUG enabled. For now I managed to circumvent this by reducing the amount of data logged.

**To Reproduce**
Steps to reproduce the behavior:
1. Log a lot of image data
2. Save recording somewegere
3. Crash

**Expected behavior**
The recording to be saved and rerun not crashing.

**Backtrace**
Not available.

**Desktop (please complete the following information):**
 - OS: Pop!_OS 22.04

**Rerun version**
0.17.0

Wumpf, 2024-07-25:
very strange, there should be at least a backtrace logged. Does anything of interest come up with `RUST_LOG=debug`
Do you know whether the viewer already hit the [memory limit](https://rerun.io/docs/howto/limit-ram) and started removing data?

teh-cmc, 2024-07-25:
You're likely running out of memory, the viewer allocates a bunch during save:
- https://github.com/rerun-io/rerun/issues/6984