rerun-io/rerun #7026: Formalize component joins that are happening within visualizers

Wumpf, 2024-07-31:
There's various kinds of joins on how we can (and do) join components within an archetype. This happens ad-hoc in visualizers and is only sparsely documented.

- Batch-joined. I.e. Arch[index] = (C1[index], C2[index], ...). Example: Points
- Instance-joined. I.e. Arch[0] = (C1[:], C2[:], ...). Example: Mesh3D
- Product-joined. I.e. Arch[i1, i2] = (C[i1], C[i2]).

`Mesh3D` & `Asset3D` are a hybrid of Instance-joined components Vertex / Indices with Product-joined component of `LeafTransform3D`.

Formalizing this would lead more easily to
* better documentation
* better ui
* consistent user queries

Loosely related to:
* https://github.com/rerun-io/rerun/issues/5259
* https://github.com/rerun-io/rerun/issues/6910

Wumpf, 2024-12-17:
Another example of this are graph nodes!
(cc: @grtlr )

teh-cmc, 2024-12-17:
Also very related to materialized aggregated queries as compute graphs.