rerun-io/rerun #7066: Dataframe view: add an `Index` column for instances and support un/collapsing

abey79, 2024-08-06:
- Un/collapsing by clicking on the +
- Collapsing by clicking on the vertical line

[image]

abey79, 2024-08-06:
@gavrelina I like the look of the `[+]` / `[–]` thing, but shouldn't we be using the same triangle as everywhere else?

gavrelina, 2024-08-26:
> @gavrelina I like the look of the `[+]` / `[–]` thing, but shouldn't we be using the same triangle as everywhere else?

You're right. 
Using ▶ and ▼ would make it more consistent. 

Initially [+] [-] was there to prevent repeating similar icons for 'sorting' and possible similarities with the 'static/temporal' components. But neither of those is used now (!! Except the sorting icon for [Chunk store browser](https://github.com/rerun-io/rerun/pull/7226), which we shall also change to [image]), hence it's not justified.

Screenshot with ▶ and ▼ for visibility: 
[image]

gavrelina, 2024-08-30:
New proposal: 

Components are grouped by entities (before we can group them by archetypes through tagged components)
- Each of the instance could be "uncollapsed" until the last index of its entity. 
- One can uncollapse by pressing an icon, which appears only on hover
- Simultaneously the other instances are uncollapsed as well until the index of the initiated item. I.e. compA[0-5], compB[0-15], if I uncollapse via compA I will see CompA[0,1,2,3,4,5] and for CompB[0,1,2,3,4,5]+10 more. 
- If there are more instances for certain component we display "N more" with ability to uncollapse further. 
- When uncollapsed — each of the rows serves as a way to collapse everything back to a "summary row".
- When uncollapsed : we still see a summary row with the total number of instances for each component.
- If one scrolls down to uncollapsed instance batch that is longer then the rest -> we explore the sticky behavior for the latest known values for other components. 
- have a separate column with (shared) indices => to help users see which [N] is a certain value they are looking at instead of calculating (is it [61?] or [62?]).
[image]

Interactions:
https://github.com/user-attachments/assets/2c6ef858-9aff-4181-ad96-d7fe15504adc

abey79, 2024-09-12:
@gavrelina I'll do that in a follow-up PR, should be too back. Tracking this here:
- https://github.com/rerun-io/rerun/issues/7412

PR #7400 (2024-09-17): Add support to expand rows to show instances in the dataframe view
### What

- Closes #7066

This PR introduces the ability to expend dataframe view rows to inspect individual instances.

Note: expanding instance can potentially lead to very loooong tables. This doesn't affect performance as we only query/draw what is visible on screen, it means we reaching the limits of egui's `f32` numerical precision. This can lead to uneven scrolling or even visual glitches if the number of instance is >>1M.

Follow-up todo:
- https://github.com/rerun-io/rerun/issues/7412

https://github.com/user-attachmen [... cut]
commit 16cc2537 (2024-09-17): Add support to expand rows to show instances in the dataframe view (#7400)

### What

- Closes #7066

This PR introduces the ability to expend dataframe view rows to inspect
individual instances.

Note: expanding instance can potentially lead to very loooong tables.
This doesn't affect performance as we only query/draw what is visible on
screen, it means we reaching the limits of egui's `f32` numerical
precision. This can lead to uneven scrolling or even visual glitches if
the number of instance is >>1M.

Follow-up todo:
- https://github.com/rerun-io/rerun/issues/7412

https://github.com/user- [... cut]
closed 2024-09-17 by abey79