rerun-io/rerun #7068: Dataframe view: deduplicate row with identical value

abey79, 2024-08-06:
The dataframe view has a lot of duplicated content, especially in time range mode (same entity for many timestamps, etc.).

- deduplicate content of the time and/or entity path columns (ie. write the content only once
- add stroke lines to make cell "merged" state more explicit
- "stick" the cell value to the top of the table when scrolling

[image]

gavrelina, 2024-08-30:
if we start with an entity on top -> most likely this behavior will not happen as often (only if smth was logged at the same time tick multiple times, which is not a majority of cases). Because normally there will be 1 row per a time tick. Hence this will be useful after enabling changing 'Entity as a row' VS 'Entity as a column'? 

https://github.com/user-attachments/assets/c58b36a0-4729-4479-bb48-090d85e3993a

abey79, 2024-09-09:
Probably only useful with:
- https://github.com/rerun-io/rerun/issues/7379