rerun-io/rerun #7070: Dataframe view: sorting and primary column selection

abey79, 2024-08-06:
In time range mode, there are two "header" columns: time and entity path. It should be possible to:
- select which one is the "primary" one (eg. appears first and is used for grouping)
- choose the sort direction independently for both of them...
- ...using an icon in the column header.

Currently, it's possible to choose the primary column, but sort order is not indepenent.

Here, time is primary, and both are sorted in ascending direction:
[image]

gavrelina, 2024-09-11:
will leave a proposal here to not forget it! 

https://github.com/user-attachments/assets/241820f3-88cc-409d-91d4-c27ab98846b9

abey79, 2025-08-29:
Since we probably don't want #7379, we likely don't want that either.

closed 2025-08-29 by abey79