rerun-io/rerun #7081: Container doesn't take ownership as expected (borrows from temporary objects)

facontidavide, 2024-08-06:
**Describe the bug**

I have a class like this:

```c++
rerun::Boxes2D createBoxes(....) {
  std::vector<rerun::Position2D> centers;
  std::vector<rerun::datatypes::Vec2D> sizes;
  // fill the vectors
  return rerun::Boxes2D::from_centers_and_sizes(centers, sizes);
}
```

Unfortunately, the result seems to be corrupted, ore specifically the centers (half-sizes, seems to be ok).
I believe that the problem is that the `Collection` is borrowing the reference to the centers, instead of taking owership.

The following code fixes the issue:

```c++
rerun::Boxes2D createBoxes(....) {
  std::vector<rerun::Position2D> centers;
  std::vector<rerun::datatypes::Vec2D> sizes;
  // fill the vectors
  auto boxes rerun::Boxes2D::from_sizes(sizes);
  boxes.centers = rerun::Collection<rerun::Position2D>::take_ownership(std::move(centers));
  return boxes;
}
```

**To Reproduce**
Try the function mentioned above and then print the centers:

```c++
auto boxes = risc::createBoxes(....);
for (const auto& center : *(boxes.centers)) {
  std::cout << center.x() << " / " << center.y() << "\n";
}
```

**Rerun version**
0.17.0

Wumpf, 2024-08-06:
This is pretty much by design and document as such (although maybe not easy to find :/).
`rerun::Collection` tries to borrow whenever possible to avoid copies. The way this works is that everything that isn't created from a temporary (r value reference) will go down the `borrow` path of the collection.

That said, I'm starting to regret this design. It's great to have the _option_ of borrowing data, but we do it too aggressively. This isn't Rust where we can do this kind of operation safely, instead this makes everything very easy to fail as you've experienced.
I think we should change `rerun::Container` to borrow only on explicit codepath. This will be a breaking change for the container adapter system which right now is based on the premise of behaving like this with arbitrary types - instead we need to restyle this to have adapters that interact with explicit `borrow` methods.

Somewhat related to this recent change [here](https://github.com/rerun-io/rerun/pull/7055) where I made it easier to do explicit borrow & ownership taking.

facontidavide, 2024-08-07:
Yes, I can understand that you are trying to import good ideas from Rust, but without the compiler support, you CANNOT do that.

My suggestion is to make the C++ API more idiomatic, using what people will understand in an non ambiguous way.

```cpp
// zero-cost move semantic. Probably what I would propose by default in the tutorials
Collection(std::vector<T>&& vect)

// Reference to extenal object. Everyone understands that you are NOT taking owership of this object and that they are
// responsible for its life-time
Collection(const std::vector<T>* vect)

// Shared ownership. Safe in pretty much all the cases, but a bit more cumbersome on the caller side
Collection(std::shared_ptr<std::vector<T>> vect);

// What shall we do here? The only real safe options is to make a copy
Collection(const std::vector<T>& vect);
```

facontidavide, 2024-08-07:
``` cpp
    template <typename TElement>
    inline Collection<TElement> borrow(const std::vector<TElement>& data) {
        return Collection<TElement>::borrow(data);
    }
```

This is a recipe for disaster. Here the only sane thing to do is to take ownership.

Suggestion

``` cpp
    template <typename TElement>
    inline Collection<TElement> borrow(const std::vector<TElement>& data) {
        return Collection<TElement>::take_ownership(data);
    }

    template <typename TElement>
    inline Collection<TElement> borrow(std::vector<TElement>&& data) {
        return Collection<TElement>::take_ownership(std::move(data));
    }
    
    template <typename TElement>
    inline Collection<TElement> borrow(const std::vector<TElement>* data) {
        return Collection<TElement>::borrow(*borrow);
    }
```

Wumpf, 2024-08-07:
I'm not entirely sold on the idea that passing a pointer signifies borrowing data of the underlying vector well enough. Since rerun collection takes internally a pointer to `vect.data()` this still suffers from iterator-invalidation (i.e. resize of the vector break the collection). I'd rather make it all the way explicit.
Passing a pointer to the raw data itself == borrow makes more sense for me though as there's then no obvious way of deallocating the vector.

Also note that these two
```cpp
Collection(std::vector<T>&& vect);
Collection(const std::vector<T>& vect);
```
are more or less redundant to each other an can be replaced with 
```
Collection(std::vector<T> vect);
```
which then internally always moves.
Rationale:
* If an r-value reference is passed the vector is moved twice (cheap!)
* If a reference/l-value is passed the vector is copied (unavoidable) and then moved (cheap).

We use that pattern in a lot of places already to cull down on the number of constructors/builder methods.

Wumpf, 2024-08-07:
For posterity and in defense of how `rerun::Collection` eagerly borrows: The archetype types were not meant as a storage object in any of the SDKs and should ideally only ever exist as a temporary in a log call. Then the entire chain works out since user data gets adapter on the fly to a collection which then is temporarily stored in the archetype which then is logged/serialized.
Practically speaking though that's .. often impractical 😅 beyond simple samples.

facontidavide, 2024-08-07:
> Practically speaking though that's .. often impractical 😅 beyond simple samples.

Yes, this is why I started creating this helper functions to make the code nicer :wink: 

You are correct about the fact that `Collection(std::vector<T> vect)` may kill two birds with a stone.

The constructor 
```
explicit Collection(const std::vector<T>* vect);
```
Are intentionally "ugly" because most C++ developer will look at that and a big red flag will light in their brain.

That is specifically what I want :smile:

emilk, 2024-08-08:
The "pointer-means-borrowing is a neat idea, but it leaves the problem of a user now being able to pass in `NULL/nullptr`, but I guess we can just define that as meaning "empty collection".

closed 2025-01-23 by Wumpf