rerun-io/rerun #7115: Rerun viewer becomes slow when there are several hundreds of entities

emilk, 2024-08-08:
We currently don't have a good example of this. The best we got is

`pixi run py-build-examples && pr -e examples open_photogrammetry_format --no-frames`

and maybe

* https://github.com/rerun-io/rerun/pull/5449

jleibs, 2024-10-17:
Here's a very simple example that makes the viewer totally unuable:
```python
import rerun as rr

from rerun.datatypes import RotationAxisAngle

rr.init("rerun_example_many_entities", spawn=True)

def one_strip(number_points: int):
    return rr.LineStrips3D(
        [[0.3 if i % 2 == 0 else -0.3, i, 0] for i in range(number_points)],
    )

def rotate(theta: float):
    return rr.Transform3D(
        rotation_axis_angle=RotationAxisAngle(axis=[0, 0, 1], degrees=theta),
        axis_length=0,
    )

def draw_lines(lines_count: int, pts_per_line: int):
    theta_delta = 360 / lines_count
    for i in range(lines_count):
        rr.log(f"line_{i}", rotate(theta_delta * i))
        rr.log(f"line_{i}/strip", one_strip(pts_per_line))

draw_lines(5000, 10)
```

joelreymont, 2025-10-31:
Took a stab with PR #11743.