rerun-io/rerun #7134: Rust API examples of DepthImage and Image are not working

makeecat, 2024-08-09:
**Describe the bug**
Rust examples of DepthImage and Image are not working
**To Reproduce**
Steps to reproduce the behavior:
build the DepthImage example from https://rerun.io/docs/reference/types/archetypes/depth_image
build the Image example from https://rerun.io/docs/reference/types/archetypes/image
**Expected behavior**

The example of DepthImage cannot compile
**Backtrace**
```
error[E0277]: the trait bound `rerun::TensorData: From<ArrayBase<OwnedRepr<u16>, Dim<[usize; 2]>>>` is not satisfied
  --> src/main.rs:12:23
   |
12 |     let depth_image = rerun::DepthImage::try_from(image)?.with_meter(10_000.0);
   |                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `From<ArrayBase<OwnedRepr<u16>, Dim<[usize; 2]>>>` is not implemented for `rerun::TensorData`, which is required by `rerun::TensorData: TryFrom<ArrayBase<OwnedRepr<u16>, Dim<[usize; 2]>>>`
   |
   = help: the following other types implement trait `From<T>`:
             <rerun::TensorData as From<&[f32]>>
             <rerun::TensorData as From<&[f64]>>
             <rerun::TensorData as From<&[i16]>>
             <rerun::TensorData as From<&[i32]>>
             <rerun::TensorData as From<&[i64]>>
             <rerun::TensorData as From<&[i8]>>
             <rerun::TensorData as From<&[rerun::external::arrow2::types::f16]>>
             <rerun::TensorData as From<&[u16]>>
           and 14 others
   = note: required for `ArrayBase<OwnedRepr<u16>, Dim<[usize; 2]>>` to implement `Into<rerun::TensorData>`
   = note: required for `rerun::TensorData` to implement `TryFrom<ArrayBase<OwnedRepr<u16>, Dim<[usize; 2]>>>`

For more information about this error, try `rustc --explain E0277`.
error: could not compile `depth_rerun` (bin "depth_rerun") due to 1 previous error
```
**Desktop (please complete the following information):**
 - OS:  macOS 15.0

**Rerun version**
0.17.0

makeecat, 2024-08-09:
Can someone provide rust example of logging image and depth image with the current API of 0.17.0?

emilk, 2024-08-12:
The given example should work, but you need to use the same version of `ndarray` as Rerun was built for, which is `0.15`. I see there is a 0.16 out now, so I suspect that's the problem. Change your import to `ndarray = "0.15"` and it should work!

closed 2024-08-12 by emilk

makeecat, 2024-08-30:
Is it possible for you to provide an example to construct depth image without using ndarary? Can you show an example with manipulation of TensorData?