rerun-io/rerun #7138: Save dialog doesn't work on Web

emilk, 2024-08-10:
Rerun menu -> Save

[image]

->

jprochazk, 2024-08-12:
It works if you click on the `click here to download your file` text

emilk, 2024-08-12:
Oh no… ok, that confused the hell out of me. We should make a PR to `rfd` to improve this.

What do you think about the possibility of being able to directly start the file download on the first click – what would be required for that?

jprochazk, 2024-08-13:
> What do you think about the possibility of being able to directly start the file download on the first click – what would be required for that?

There was some PR in progress for that:
- https://github.com/PolyMeilex/rfd/pull/129

Looks like it stalled though

jprochazk, 2024-08-13:
Posted my thoughts in the original issue: https://github.com/PolyMeilex/rfd/issues/128#issuecomment-2286581683

TL;DR: We could use better APIs in Chromium-based browsers, and otherwise fall back to the existing implementation