rerun-io/rerun #7145: rerun not working

PriyangshuMazumder, 2024-08-12:
import rerun as rr
import cv2
from __future__ import annotations

import argparse
import json
import logging
import os
from dataclasses import dataclass
from pathlib import Path
from typing import Any, Final, Sequence

import cv2
import numpy as np
import numpy.typing as npt
import requests
import rerun as rr  # pip install rerun-sdk
from PIL import Image

VIDEO_CODEC = "mp4v"

cap = cv2.VideoCapture('/Users/priyangshumazumder/Desktop/DeepFaceExp/conv_input.mp4')
height = int(cap.get(cv2.CAP_PROP_FRAME_HEIGHT))
width = int(cap.get(cv2.CAP_PROP_FRAME_WIDTH))
fps = cap.get(cv2.CAP_PROP_FPS)
n_frames = int(cap.get(cv2.CAP_PROP_FRAME_COUNT))

print(f'Height {height}, Width {width}')
print(f'FPS : {fps:0.2f}')

# for frame in tqdm(range(n_frames), total=n_frames):
for frame in range(n_frames):
    ret, bgr = cap.read()
    rr.set_time_sequence("frame", frame)
    if not ret:
        logging.info("End of video")
        break
    rgb = cv2.cvtColor(bgr, cv2.COLOR_BGR2RGB)
    rr.log(
        "image",
        rr.Image(rgb)
    )

for this code it shows
 ---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[6], [line 36](vscode-notebook-cell:?execution_count=6&line=36)
     [34](vscode-notebook-cell:?execution_count=6&line=34) for frame in range(n_frames):
     [35](vscode-notebook-cell:?execution_count=6&line=35)     ret, bgr = cap.read()
---> [36](vscode-notebook-cell:?execution_count=6&line=36)     rr.set_time_sequence("frame", frame)
     [37](vscode-notebook-cell:?execution_count=6&line=37)     if not ret:
     [38](vscode-notebook-cell:?execution_count=6&line=38)         logging.info("End of video")

AttributeError: module 'rerun' has no attribute 'set_time_sequence'

PriyangshuMazumder, 2024-08-12:
installed as pip3 install rerun-sdk in python

Wumpf, 2024-08-12:
Please use code tags to make your code more legible and take care of picking a more descriptive title in the future (I took care of both now).

This error is most likely since you previously installed the `rerun` python package, as opposed to `rerun-sdk`. Try `pip3 uninstall rerun` and then run again.

I also noticed that your code snippet is missing a call to `rr.init` and therefore tries to run `rr.set_time_sequence` without setting up a recording (that is **not** the source of the error your seeing, but will cause you issues once you solved that)

closed 2024-08-12 by emilk