rerun-io/rerun #7167: Support columns with variable partitions in Rust's `send_column` api

Wumpf, 2024-08-13:
Both Python & C++ APIs have ways of partitioning a component column arbitrarily to e.g. log point clouds in a single `send_columns` call.
Rust however, right now only supports partitions of length 1 for each element (i.e. at each timestep you end up with an array of length 1) which is still great for the usecase of sending e.g. `Scalar`, but not the full picture.

Note that the lower level `send_chunk` API that the Rust recording stream comes with has none such restrictions (and is what is used by Python and C++).

teh-cmc, 2024-10-22:
If it's not outright blocked by, it would at the very least _heavily_ benefit from:
* https://github.com/rerun-io/rerun/issues/7245

teh-cmc, 2024-12-03:
Somewhat related, the Rust definition of `send_columns` seems ~wrong~ not optimal:
```rust
    pub fn send_columns<'a>(
        &self,
        ent_path: impl Into<EntityPath>,
        timelines: impl IntoIterator<Item = TimeColumn>,
        components: impl IntoIterator<Item = &'a dyn re_types_core::ComponentBatch>,
    ) -> RecordingStreamResult<()> {
```

Why take an iterator of `ComponentBatch`es instead of the more flexible `AsComponents`?

closed 2025-01-21 by teh-cmc