rerun-io/rerun #7177: Fonts broken with rerun 0.17 on Windows

andresv, 2024-08-14:
**Describe the bug**
Fonts are unreadable and mangled on Windows build. Look https://github.com/andresv/rerun-win-fonts

**To Reproduce**
Here is a minimal app that can be used to reproduce this issue: https://github.com/andresv/rerun-win-fonts

**Screenshots**
[image]

**Desktop**
Windows 10 Pro 19045.3930 VM running under Parallels 19.4.0, host is macOS Sonoma 14.5.
Also tried to build under github actions and it looked the same.

**Rerun version**
`0.17`

andresv, 2024-08-14:
I sent exe to my friend who has win11 and it runs fine in there.

emilk, 2024-08-14:
This looks like a graphics driver issue to me. Have you perhaps changed the default settings for them?

andresv, 2024-08-14:
Tried with normal rerun and it is the same:

[image]

I haven't tweaked windows VM graphics settings.