rerun-io/rerun #7191: LineStrips2D radii not working as expected for real world data

Famok, 2024-08-15:
**Describe the bug**
The radii of Linestrips2D does not what I expect it to do, which is to increase the line width.
Instead it only increases the width of each segment, which will lead to a weird behaviour if you have real world data (e.g. with noise).

**To Reproduce**
Code to show the issue:
```
import rerun as rr
import numpy as np

x = np.linspace(0,2*np.pi,1000)
y = np.random.rand(len(x))*0.1

xy = np.array([x,y]).T

rr.init('testing', recording_id= ' test', spawn = True)
for r in [0.001, 1]:
    rr.log(
        f"random line {r=}",
        rr.LineStrips2D(
            [xy],
            radii=r
        ),
    )

```

**Expected behavior**
I was expecting the line as a whole to get wider (similar how this works in powerpoint for example).

**Desktop (please complete the following information):**
Windows 11 23H2

**Rerun version**
rerun_py 0.18.0-alpha.1+dev [rustc 1.76.0 (07dca489a 2024-02-04), LLVM 17.0.6] x86_64-pc-windows-msvc main 9e50d2c, built 2024-08-14T13:15:39Z

nikolausWest, 2024-08-15:
Thanks for the issue @Famok!

Could you add some screenshots comparing what you're seeing in Rerun vs the behavior you're expecting (powerpoint)? We're having a little trouble interpreting exactly what you mean

Wumpf, 2024-08-15:
took me a bit indeed but I think I see what you mean now, the line radius behaves strangely when it goes about individual line lengths:

https://github.com/user-attachments/assets/c9d54e35-a0d4-4321-88a0-b425f6891cc8

Seems to be closely related to:
* https://github.com/rerun-io/rerun/issues/829

abey79, 2024-08-15:
Another depiction. Both are the same line. The yellow one has _very_ thick segments. The "effect" is due to the lack of (miter/round) join handling.

[image]

Famok, 2024-08-15:
@abey79 this is what I mean. For some reason I'm unable to upload  images myself so I could show how my expected behavior looks like.

closed 2024-11-12 by Wumpf