rerun-io/rerun #7194: Add Stream to existing spaceview

Famok, 2024-08-15:
**Is your feature request related to a problem? Please describe.**
Lets say I have two features sensor/feature1 and sensor/feature2 and I want them in the same chart, I have to add sensor to a new space view. But what if I added sensor/feature1 and then decided that I want sensor/feature2 in there as well?
--> I have to make a new spaceview

**Describe the solution you'd like**
An addition to the right-click menu of the streams with the option to add to existing space view. This could also allow me to combine data from different parents (e.g. sensor1/feature3 and sensor3/feature10)

Wumpf, 2024-08-15:
Context menu for adding to existing spaces makes sense to me. Drag an drop would also be nice (issue about this [here](https://github.com/rerun-io/rerun/issues/7108).
But generally this is already available via the `contents` of a view. They are are also accessible from blueprint

https://github.com/user-attachments/assets/49a7cea5-7d39-40a7-88dc-6c2f2ef1875d

Is that what you meant?

Famok, 2024-08-15:
@Wumpf Yeah exactly. That was hidden in plain sight, I was trying to drag the streams, right click them, right click the views but didn't think of looking on the right panel. Thanks alot, that helps for now.