rerun-io/rerun #7195: Spurious error with `Colormap` data UI

abey79, 2024-08-15:
The colormap component UI can trigger a spurious error in some circumstances.

[image]

Error:
```
[2024-08-15T07:57:45Z ERROR re_viewer_context::gpu_bridge::colormap] Failed to paint colormap preview: Condition failed: `resolution_in_pixel[0] > 0 && resolution_in_pixel[1] > 0`
```

Repro:
- run `check_all_components_ui.py`
- select `all_components_single`
- randomly scroll the selection panel vertically until the error shows up (probably when the colormap is rendered just offscreen)

code:
https://github.com/rerun-io/rerun/blob/819e021be96f633380bf711fc3fdc5267f6232da/crates/viewer/re_viewer_context/src/gpu_bridge/mod.rs#L160-L168

abey79, 2024-08-15:
The fix might be as easy as changing this `ensure!` line to an early exit, but I'm not sure if there would be hidden implications.

Wumpf, 2024-08-15:
I wonder what causes this to happen. But it an early out here instead seems safe enough to me

abey79, 2024-08-15:
My bet is a situation where the `ListItem` is _just_ visible on screen, so it draws its content, but the gradient is fully offscreen (because it's slightly smaller than the `ListItem`).