rerun-io/rerun #7200: Individual Timeline Points get lost when using send_columns

Famok, 2024-08-15:
**Describe the bug**
When sending (scalar) values via the send_columns the timeline might stop showing individual points and only a line.
Logging the same values using set_time_seconds and log Scalar will show these.

**To Reproduce**
```
import rerun as rr
import numpy as np

rr.init('test',spawn=True)

times = np.linspace(0,1,10000)
y = np.random.rand(len(times))

rr.send_columns(
                                f'myval',
                                times=[rr.TimeSecondsColumn("sensor_time", times)],
                                components=[rr.components.ScalarBatch(y)],
                            )

# And now the same using normal logcalls
for i,t in enumerate(times):
    rr.set_time_seconds('sensor_time',t)
    rr.log('anotherval', rr.Scalar(y[i]))
```

**Expected behavior**
I expect the timeline to look the same as with normal log calls

**Screenshots**
Would love to, but seems like I can't upload screenshots

**Desktop (please complete the following information):**
Windows 11 23H2

**Rerun version**
rerun_py 0.18.0-alpha.1+dev [rustc 1.76.0 (07dca489a 2024-02-04), LLVM 17.0.6] x86_64-pc-windows-msvc main 9e50d2c, built 2024-08-14T13:15:39Z

emilk, 2024-08-15:
[image]

[image]

jprochazk, 2024-08-15:
We recently changed how the timeline retrieves the data used to render those density graphs to be done on demand in the UI, which resulted in a large improvement in ingestion times.
But because we're now doing more a bit more work on each frame, we no longer render individual time points at all times, instead we only render individual points if the total number of events within the chunk is under a threshold. With 10000 individual events in a single chunk, you're exactly at that threshold:

https://github.com/rerun-io/rerun/blob/main/crates/viewer/re_time_panel/src/data_density_graph.rs#L478

We could perform a range query of the visible time range for each sorted chunk, and then test _that_ against the threshold, instead of the full chunk as we're currently doing. With that, zooming in on a large sorted chunk would still display individual events.