rerun-io/rerun #7216: rerun-cli fails to install due to error in `time` dependency

tbillington, 2024-08-16:
```sh
$ cargo install rerun-cli --locked
...
error[E0282]: type annotations needed for `Box<_>`
  --> /home/choc/.cargo/registry/src/index.crates.io-6f17d22bba15001f/time-0.3.21/src/format_description/parse/mod.rs:83:9
   |
83 |     let items = format_items
   |         ^^^^^
...
86 |     Ok(items.into())
   |              ---- type must be known at this point
   |
help: consider giving `items` an explicit type, where the placeholders `_` are specified
   |
83 |     let items: Box<_> = format_items
   |              ++++++++

For more information about this error, try `rustc --explain E0282`.
error: could not compile `time` (lib) due to 1 previous error
warning: build failed, waiting for other jobs to finish...
error: failed to compile `rerun-cli v0.18.0`, intermediate artifacts can be found at `/tmp/cargo-installiAlqbN`.
To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.
```

```sh
$ rustc --version --verbose
rustc 1.80.0 (051478957 2024-07-21)
binary: rustc
commit-hash: 051478957371ee0084a7c0913941d2a8c4757bb9
commit-date: 2024-07-21
host: x86_64-unknown-linux-gnu
release: 1.80.0
LLVM version: 18.1.7
```

Wumpf, 2024-08-16:
I'm still trying to understand where this problem comes from (repros fine for me as well!), but meanwhile the following two commands seem to work:
* `cargo install rerun-sdk` (without `--locked`)
* `cargo binstall rerun-sdk`, that's using [`cargo binstall`](https://github.com/cargo-bins/cargo-binstall) to download & install prebuilt binaries

Wumpf, 2024-08-16:
building without `--locked` picks 0.3.36 for `time`, but given that we also use `--locked` on ci for testing it's a bit unclear to me why this isn't an issue always. Maybe because we're using an older rustc version on CI? Doesn't make sense to me

kpreid, 2024-08-16:
>  Maybe because we're using an older rustc version in the repo (via `rust-toolchain` file)?

Exacty. This error only happens with Rust ≥ 1.80, but `rust-toolchain` has 1.76. `rust-toolchain` applies to development and CI, but not people running `cargo install` from an unrelated directory.

Wumpf, 2024-08-17:
But newer rustc versions should never fail to compile crates that compile fine with older rustc versions, right? I mean if this is about the rust version selected then what we're seeing is that `time-0.3.21` compiles with rust 1.76, but not with rust 1.80 which would imply a compiler bug.

kpreid, 2024-08-17:
In general, **adding a trait implementation**, such that there are two possibilities where there used to be one, can cause this error, but nobody wants to live in the world where “you can't add more trait implementations without a major version bump”, so It's currently in the category of “possibly acceptable breakage”. Arguably, this is a design flaw in Rust (there shouldn't be a special rule for unique implementations, so the type annotation should be required always), but a lot of stuff actually relies on that rule; [here's some discussion on maybe fixing it cleverly](https://rust-lang.zulipchat.com/#narrow/stream/213817-t-lang/topic/.22only.20one.20possible.20impl.22.20inference).

closed 2024-08-19 by emilk