rerun-io/rerun #7227: Intra-timestamps join broken for spatial views on 0.18

teh-cmc, 2024-08-18:
See e.g. this:
```python
import rerun as rr

rr.init("rerun_example_points3d", spawn=True)

rr.set_time_sequence("frame", 42)
rr.log("points", rr.Points3D([[0, 0, 0], [1, 1, 1]], radii=-5, colors=[255, 0, 0]))
rr.log(
    "points", rr.Points3D([[10, 10, 10], [11, 11, 11]], radii=-5, colors=[0, 255, 0])
)
```

On 0.17 this yields what you'd expect:
[image]

On 0.18 this will fail though:
[image]

The is a regression following https://github.com/rerun-io/rerun/pull/7199, which fixes that static/override case but breaks the common case.

TODO: I have a literally jump on a plane, i'll add more notes after

Wumpf, 2024-08-18:
urgh. I feared something like that would happen but we assumed all is well when we tested overriding static data :(

teh-cmc, 2024-08-19:
I've managed to reproduce in both Python and Rust.

Fun fact: this only happens when _not_ logging to a custom timeline.

Python:
```python
import numpy as np
import rerun as rr
import time

rr.init("rerun_example_image", spawn=True)

for i in range(0, 100):
    # NOTE: uncomment this line to fix the perf issue.
    # rr.set_time_sequence("frame", i)

    for j in range(0, 100):
        image = np.zeros((200, 300, 3), dtype=np.uint8)
        image[:, :, 0] = (i + j) % 255
        image[50:150, 50:150] = (0, 255 - ((i + j) % 255), 0)

        rr.log(f"images/{j}", rr.Image(image))

    time.sleep(0.010)
```

Rust:
```rust
use ndarray::{s, Array, ShapeBuilder};

fn main() -> Result<(), Box<dyn std::error::Error>> {
    let rec = rerun::RecordingStreamBuilder::new("rerun_example_image").spawn()?;

    for i in 0..100 {
        // NOTE: uncomment this line to fix the perf issue.
        // rec.set_time_sequence("frame", i);

        for j in 0..100 {
            let mut image = Array::<u8, _>::zeros((200, 300, 3).f());
            image.slice_mut(s![.., .., 0]).fill((i + j) % 255);
            image
                .slice_mut(s![50..150, 50..150, 1])
                .fill(255 - ((i + j) % 255));

            rec.log(
                format!("images/{j}"),
                &rerun::Image::from_color_model_and_tensor(rerun::ColorModel::RGB, image)?,
            )?;
        }

        std::thread::sleep(std::time::Duration::from_millis(10));
    }

    Ok(())
}
```

teh-cmc, 2024-08-20:
Ha dang, this might be worse than I thought -- I think this breaks `log_components` workflow completely.

I'll add dedicated checklists so this doesn't surprise us again.

teh-cmc, 2024-08-28:
Found _a lot_ more issues of that nature, many of which were not introduced by https://github.com/rerun-io/rerun/pull/7199.

I'll have a PR ready for all known visualizer-query bugs, at some point.

The first problem is that the basic overrides test suite doesn't even run on 0.18.0 it turns out:
* https://github.com/rerun-io/rerun/issues/7294

PR #7297 (2024-08-28): Fix and test all known `HybridResults` issues from 0.18
This fixes all known bugs/regressions introduced in the visualizer-query layer (aka `HybridResults`) in 0.18, and adds related checklists for futureproof-ness.

As always, this is "just" a matter of carefully overriding row indices when merging data from multiple sources with different priorities.

- Fixes #7227  
- Partially reverts #7199 

TODO:
- [x] check_latest_at_partial_updates passing
- [x] check_latest_at_partial_updates comparison with 0.17
- [x] check_range_partial_updates passing
- [x] check_range_partial_updates comparison wi [... cut]
commit 9aea808c (2024-08-28): Fix and test all known `HybridResults` issues from 0.18 (#7297)

This fixes all known bugs/regressions introduced in the visualizer-query
layer (aka `HybridResults`) in 0.18, and adds related checklists for
futureproof-ness.

As always, this is "just" a matter of carefully overriding row indices
when merging data from multiple sources with different priorities.

- Fixes #7227  
- Partially reverts #7199
closed 2024-08-28 by teh-cmc