rerun-io/rerun #7237: Get local websocket port upon launching rr.serve with  ws_port=0

carueda, 2024-08-19:
**Is your feature request related to a problem? Please describe.**

I have a "server" app (in python) that dynamically launches (and kills) rerun websocket server instances. Client apps (with Rerun viewer in them) will then connect to any of those instances. So far so good. However, having to manually (pre)determine the ports those instances will run on is a bit of a hassle.

**Describe the solution you'd like**

A convenient/reasonable solution is that, upon launching with `rr.serve(ws_port=0, ...)`, there's a Rerun API that reports the local port that was assigned. (apologies if such thing already exists and I'm just not finding it.)

I'm using rerun-sdk 0.16.1.

**Describe alternatives you've considered**

Such port shows up as part of INFO logging, eg:

```
[2024-08-19T23:21:53Z INFO  re_ws_comms::server] Hosting a WebSocket server on ws://localhost:34707. You can connect to this with ...
```

but not ideal having to parse it from there.

carueda, 2025-03-06:
Hi @Wumpf - any updates on this?

I'm in the process of upgrading an application from using Rerun 0.16.1 to the latest (at this point, 0.22.1).  This is advancing well, but I note that my workaround above to get the port upon launching `rr.serve_web(ws_port=0, ...)` is not effective anymore. Thanks.

carueda, 2025-03-06:
Ah, seems like the default logging level is now less than INFO, so it should be an easy fix on my side to keep the workaround functional: just define the env var `RUST_LOG=info` for when my python program launches `serve_web`:

```
$ RUST_LOG=info poetry run python rrws/main.py
[2025-03-06T23:06:24Z INFO  re_ws_comms::server] Hosting a WebSocket server on ws://localhost:42459. You can connect to this with a native viewer (`rerun ws://localhost:42459`) or the web viewer (with `?url=ws://localhost:42459`).
[2025-03-06T23:06:25Z INFO  re_sdk::web_viewer] Hosting a web-viewer at http://localhost:36621?url=ws://localhost:42459
```

carueda, 2025-03-25:
I've removed the workaround and now pre-determine the socket port before calling `serve_web`. Closing this out to help tidy up the tracker.

Wumpf, 2025-03-31:
Thanks! But that's still an annoying issue though, there should be some more standardized way to retrieve the port when using automatic port selection