rerun-io/rerun #7248: Rerun C++ SDK: undefined symbols (arch mismatch) when cross-compiling for macOS x64

rafaelspring, 2024-08-20:
**Describe the bug**
Using the rerun C++ SDK. My machine is a MacBook Air M1 (arm64). I am including rerun in an app that (for various reasons) I need to compile to x64 architecture (using CMAKE_OSX_ARCHITECTURES = x86_64).

I'm getting tons of arch mismatch messages from the linker of this kind:
```
ld: warning: ignoring file '../arrow/lib/libarrow_bundled_dependencies.a[15](init.c.o)': found architecture 'arm64', required architecture 'x86_64'
ld: warning: ignoring file '../arrow/lib/libarrow_bundled_dependencies.a[14](options.c.o)': found architecture 'arm64', required architecture 'x86_64'
ld: warning: ignoring file '../arrow/lib/libarrow_bundled_dependencies.a[13](heap.c.o)': found architecture 'arm64', required architecture 'x86_64'
...
```

then further down many undefined symbols
```
Undefined symbols for architecture x86_64:
  "arrow::ExportType(arrow::DataType const&, ArrowSchema*)", referenced from:
      rerun::ComponentType::register_component() const in librerun_sdk.a[78](component_type.cpp.o)
  "arrow::ExportArray(arrow::Array const&, ArrowArray*, ArrowSchema*)", referenced from:
      rerun::ComponentBatch::to_c_ffi_struct(rr_component_batch&) const in librerun_sdk.a[76](component_batch.cpp.o)
      rerun::ComponentColumn::to_c_ffi_struct(rr_component_column&) const in librerun_sdk.a[77](component_column.cpp.o)
      rerun::TimeColumn::to_c_ffi_struct(rr_time_column&) const in librerun_sdk.a[188](time_column.cpp.o)
...
```

**To Reproduce**
Steps to reproduce the behavior:
* Use a Mac with arm CPU.
* Create a CMake-based project (app or shared lib) that uses the rerun C++ SDK.
* Set CMAKE_OSX_ARCHITECTURES to x86_64
* Use SDK in code.
* Build.

**Expected behavior**
Code building without errors.

**Desktop (please complete the following information):**
OS: macOS Sonoma 14.6.1

**Rerun version**
Whatever is here: https://github.com/rerun-io/rerun/releases/latest/download/rerun_cpp_sdk.zip

Wumpf, 2024-09-09:
Our CMake file is generally not set up for cross compilation and it's not something we test on CI either.
Contributions very much welcome!

cedricnew92, 2024-09-19:
I'm running the build with -DCMAKE_OSX_ARCHITECTURES="x86_64;arm64" from an intel mac, the static library generated is just x86_64
$ lipo -info libarrow_bundled_dependencies.a
Non-fat file: libarrow_bundled_dependencies.a is architecture: x86_64

Wumpf, 2024-10-22:
Related to:
* https://github.com/rerun-io/rerun/issues/7445