rerun-io/rerun #7252: DepthImage flickering in rerun 0.18

makeecat, 2024-08-22:
**Describe the bug**

The depthimage logged in rerun 0.18 may become inconsistent between frames. It looks like flickering. The same behavior does not exist in 0.17. I feel like it could be related to the background rrd file compression?

**To Reproduce**
Steps to reproduce the behavior:
1. compile and run code in https://github.com/makeecat/Peng
2. you will see depthimage flickering in the logged data
3. The rerun related code is https://github.com/makeecat/Peng/blob/a936a9dd47ecde4719b268004082aa0d06eba670/src/main.rs#L1276

**Desktop (please complete the following information):**
 - OS: macOS, Ubuntu 22.04 (x64)

**Rerun version**
0.18.0

emilk, 2024-08-23:
Can you please elaborate - inconsistent how? Do you have some screenshots or video you can add?

You're using an 8-bit buffer for the depth image - this will cause severe quantizing. You can use a full `f32` instead:

```diff
-    let depth_image_buffer: image::GrayImage =
-        image::ImageBuffer::from_fn(width as u32, height as u32, |x, y| {
-            let depth = depth_image[y as usize * width + x as usize];
-            if depth.is_infinite() {
-                image::Luma([0u8])
-            } else {
-                let normalized_depth = (depth - min_depth) / (max_depth - min_depth);
-                image::Luma([(normalized_depth * 255.0) as u8])
-            }
-        });
+    let depth_image_buffer = ndarray::Array::from_iter(depth_image.iter().cloned())
+        .into_shape((height, width))
+        .unwrap();
```

with

```toml
ndarray = "0.15"
```

in your Cargo.toml

makeecat, 2024-08-23:
I will try to use f32 or u16 illustrated in https://rerun.io/docs/reference/types/archetypes/depth_image. I am using u8 since that is the only way I can work with related to https://github.com/rerun-io/rerun/issues/7134

Here is the flickering video:
[image]

makeecat, 2024-08-23:
It could be related to https://github.com/rerun-io/rerun/issues/6106

emilk, 2024-08-26:
Yes, I think this is just  #6106 aggravated by using `u8` instead of `f32`

closed 2024-08-26 by emilk