rerun-io/rerun #7254: Memory usage increase indefinitely after window minimized on Linux

chengguizi, 2024-08-23:
**Describe the bug**

When the desktop viewer window is minimised or on some machine bring to background of other windows, there are data seemingly being cached by the rerun process.

When the viewer window is bring to foreground again, i can see it being fast forward in replay, before it goes back to realtime again.

If i leave the window minimised for too long, eventually it dries up the memory on system and hangs the system

**To Reproduce**
Steps to reproduce the behavior:
1. Run with `rerun --drop-at-latency 500ms --server-memory-limit 200MB --memory-limit 200MB`
2. Stream image data over time
3. Minimise the viewer window
4. Observer using htop to prove that memory increases steadily

**Expected behavior**

Memory usage should be bounded by `--server-memory-limit 200MB --memory-limit 200MB`

**Screenshots**

**Backtrace**

**Desktop (please complete the following information):**
 - OS: Linux 20.04 / 22.04

**Rerun version**

0.15.1 or 0.18.0 tested

**Additional context**

teh-cmc, 2024-08-23:
The viewer currently ingest data on the UI thread -- when your window is in the background, it will either slow down the UI thread massively or stop it altogether depending on your OS / window manager.

This is a known issue, see:
* https://github.com/rerun-io/rerun/issues/1960
* https://github.com/rerun-io/rerun/issues/4298

teh-cmc, 2024-08-23:
Closing as duplicate

closed 2024-08-23 by teh-cmc

chengguizi, 2024-08-23:
@teh-cmc Thanks for quick response, understand it is due to UI thread processing the ingestion. Yes would really appreciate to have this solved. Otherwise, minimise / lock screen will basically makes the whole OS OOM eventually and hangs.