rerun-io/rerun #7268: Add snapshot tests

abey79, 2024-08-26:
Snapshot tests (aka comparing screenshots to a stored reference) are great to catch regression of all sorts and we should have them.

For it to be convenient, we'd need at least:
- a way to create/test them headless on CI, ideally on all runner architectures (Linux/Mac/Windows)—but any would do for starters
- a way to highlight differences between the reference and actual that is easily accessible also for CI jobs
- an easy way to update the snapshots when they break due to changes/new features
- a way to store the snapshots that doesn't clutter the git repo (vello apparently uses git-lfs: https://github.com/linebender/vello/pull/610)

Wumpf, 2024-09-09:
(renamed ticket because we already have "snapshot tests" using the "insta" framework)

emilk, 2024-11-29:
Done

closed 2024-11-29 by emilk