rerun-io/rerun #7278: Support for logging & plotting ScalarBatch types

jleibs, 2024-08-26:


jleibs, 2024-09-10:
> I'm not sold on the name. "Signals" is often used to mean "callbacks/delegates", especially in the context of UIs.

My original thought is the signals/slots terminology was pretty removed from our particular use-case, but I went and took a look at Vega which also happens to use signals in the way you're describing (https://vega.github.io/vega/docs/signals/) as a specification related to reactive UI input events.

@emilk any alternative suggestions?

jleibs, 2024-09-10:
Other names that have come up:
 - Vector (concerns that it's too overloaded with spatial, e.g. Vec3)
 - Tensor1D (not super obvious to users that haven't heard the tensor terminology before)
 - Array (probably too generic)
 - ScalarArray :shrug:

oceanusxiv, 2024-11-17:
+1 on this feature, having a for loop to log state vectors on a timeline is really not ideal. Ideally the viewer can internally do the unpacking. My 2c is ScalarArray is not bad as a name. if the idea is that this is just a bundle of the original `Scalar` archetype.

abey79, 2025-02-03:
> Other names that have come up:
> 
> * Vector (concerns that it's too overloaded with spatial, e.g. Vec3)
> * Tensor1D (not super obvious to users that haven't heard the tensor terminology before)
> * Array (probably too generic)
> * ScalarArray 🤷

Did we consider just extending the existing `Scalar` archetype?

This just works:

```python
import rerun as rr

rr.init("rerun_example_scalars")
rr.connect_tcp()

rr.set_time_sequence("frame", 1)
rr.log("test", rr.Scalar([1, 2, 3]))

rr.set_time_sequence("frame", 2)
rr.log("test", rr.Scalar([2, 2, 3]))

rr.set_time_sequence("frame", 3)
rr.log("test", rr.Scalar([4, 2, 3]))
```

Except only the first element is displayed:

[image]

We'd "just" need to make the time series view `ScalarBatch`-aware, and add an optional `name` field to the archetype.

Wumpf, 2025-02-11:
>  and add an optional name field to the archetype.

Or alternatively just extend all properties on `SeriesLine`/`SeriesPoint` as well. Which then also solves the question of how this interacts.

A bunch of rough edges I can think of:
* plural/singular naming becomes a bit strange compared to how we usually do things
* some api might need extra attention to work as-is and incorporate arrays
* corner cases for varying array size over time (that's the case with all proposals)
    * I think we should do whatever is most performant to resolve those - we should be allowed to assume that the array size doesn't change over time

Wumpf, 2025-02-12:
Superseeded by
* https://github.com/rerun-io/rerun/issues/9005