rerun-io/rerun #7292: Performance regression with many poses

teh-cmc, 2024-08-28:
Via @Danvil in https://github.com/rerun-io/rerun/issues/7222#issuecomment-2313667330:
> With branch `master` (as of 8/27, 3PM PDT) it feels a little bit faster but performance is still lower than rerun 0.16.
> 
> I tried some bisection of the rerun visualizations in my application and the bottleneck now seems to be this code:
> 
> ```
> rr.set_time_seconds("timestamp", my_timestamp);
> rr.log("path", &rerun::Clear::recursive())?;
> for (i, pose) in path.iter().enumerate() {
>     rr.log(format!("{tag}/{i}"), &rerun::Transform3D::from_translation_mat3x3(pose.t3, pose.r3))?;
> }
> ```
> 
> (there are 250 poses)
> 
> Without this visualization `0.18` also is much faster. So it seems like a large part of the slowdown from `0.16` to `0.18` is due to visualization of the poses.

jleibs, 2024-08-28:
Any chance the recursive clear is impacting things now that we do query-time clears?

teh-cmc, 2024-08-28:
(This will be my ever-involving live-investigation comment)

---

```python
import rerun as rr

rr.init("rerun_example_transform3d")
rr.save("/tmp/buncha_transforms_without_clear.rrd")

rr.set_time_seconds("timestamp", 42)
for i in range(0, 250):
    rr.log(f"transforms/{i}", rr.Transform3D(translation=[i, 0, 0]))
```

---

```python
import rerun as rr

rr.init("rerun_example_transform3d")
rr.save("/tmp/buncha_transforms_with_clear.rrd")

rr.set_time_seconds("timestamp", 42)
rr.log("transforms", rr.Clear(recursive=True))
for i in range(0, 250):
    rr.log(f"transforms/{i}", rr.Transform3D(translation=[i, 0, 0]))
```

closed 2024-08-28 by jleibs