rerun-io/rerun #7294: check_overrides_2d.py fails on 0.18

teh-cmc, 2024-08-28:
```
$ python tests/python/release_checklist/check_overrides_2d.py

[2024-08-28T07:40:31Z INFO  re_sdk_comms::server] New SDK client connected from: 127.0.0.1:35426
/home/cmc/dev/rerun-io/rerun/venv/lib/python3.8/site-packages/rerun_sdk/rerun/_converters.py:155: FutureWarning: The input object of type 'Range1D' is an array-like implementing one of the corresponding protocols (`__array__`, `__array_interface__` or `__array_struct__`); but not a sequence (or 0-D). In the future, this object will be coerced as if it was first converted using `np.array(obj)`. To retain the old behaviour, you have to either modify the type 'Range1D', or assign to an empty array created with `np.empty(correct_shape, dtype=object)`.
  return np.asarray(data, dtype=np.float64)
Traceback (most recent call last):
  File "tests/python/release_checklist/check_overrides_2d.py", line 160, in <module>
    run(args)
  File "tests/python/release_checklist/check_overrides_2d.py", line 151, in run
    log_boxes()
  File "tests/python/release_checklist/check_overrides_2d.py", line 70, in log_boxes
    visual_bounds = rrb.VisualBounds2D(x_range=[-5.5, 5.5], y_range=[-3.0, 3.0])
  File "/home/cmc/dev/rerun-io/rerun/venv/lib/python3.8/site-packages/rerun_sdk/rerun/blueprint/archetypes/visual_bounds2d_ext.py", line 32, in __init__
    self.__attrs_init__(range=blueprint_components.VisualBounds2D(x_range=x_range, y_range=y_range))
  File "<attrs generated init rerun.blueprint.archetypes.visual_bounds2d.VisualBounds2D>", line 3, in __attrs_init__
  File "/home/cmc/dev/rerun-io/rerun/venv/lib/python3.8/site-packages/rerun_sdk/rerun/_baseclasses.py", line 195, in _required
    return cls(data)
  File "/home/cmc/dev/rerun-io/rerun/venv/lib/python3.8/site-packages/rerun_sdk/rerun/_baseclasses.py", line 181, in __init__
    array = self._native_to_pa_array(data, self._ARROW_TYPE.storage_type)
  File "/home/cmc/dev/rerun-io/rerun/venv/lib/python3.8/site-packages/rerun_sdk/rerun/datatypes/range2d.py", line 110, in _native_to_pa_array
    Range1DBatch([x.x_range for x in data]).as_arrow_array().storage,  # type: ignore[misc, arg-type]
  File "/home/cmc/dev/rerun-io/rerun/venv/lib/python3.8/site-packages/rerun_sdk/rerun/_baseclasses.py", line 181, in __init__
    array = self._native_to_pa_array(data, self._ARROW_TYPE.storage_type)
  File "/home/cmc/dev/rerun-io/rerun/venv/lib/python3.8/site-packages/rerun_sdk/rerun/datatypes/range1d.py", line 68, in _native_to_pa_array
    return Range1DExt.native_to_pa_array_override(data, data_type)
  File "/home/cmc/dev/rerun-io/rerun/venv/lib/python3.8/site-packages/rerun_sdk/rerun/datatypes/range1d_ext.py", line 23, in native_to_pa_array_override
    ranges = flat_np_float64_array_from_array_like(data, 2)
  File "/home/cmc/dev/rerun-io/rerun/venv/lib/python3.8/site-packages/rerun_sdk/rerun/_validators.py", line 59, in flat_np_float64_array_from_array_like
    array = to_np_float64(data)
  File "/home/cmc/dev/rerun-io/rerun/venv/lib/python3.8/site-packages/rerun_sdk/rerun/_converters.py", line 155, in to_np_float64
    return np.asarray(data, dtype=np.float64)
TypeError: float() argument must be a string or a number, not 'Range1D'
```

teh-cmc, 2024-08-28:
It's actually everything that uses `rrb.VisualBounds2D`

teh-cmc, 2024-08-28:
It looks like this was fixed on `main` at some point since the 0.18.0 release :question: I have no idea where or how, but it works :shrug:

closed 2024-08-28 by teh-cmc

emilk, 2024-08-28:
Let's make sure to test this on the 0.18.1 release branch